MarlinFirmware/Marlin #13094: Custom machine: Marlin set pin HIGH before reaching position

athulsnair, 2019-02-07:
Hello,
     I am trying to build a liquid dispensing machine and using Marlin to control it. Already made a small python based GUI to generate gcode and use Repetier Host to send Gcode to Controller(RAMPS 1.4).

What machine does is, One tool will make an impression in a powder, then syringe will come on top of that impression, lower itself a little and dispense liquid. I use M42 command to send pulse to Liquid dispenser to trigger it. Upon receiving the signal it should dispense liquid for a set time in liquid dispense controller. 

Marlin works fine in every condition except for the crucial part; Triggering GPIO pin

Machine is a modified CNC. Z axis and E0 is placed on the Gantry. Z axis carries Syringe & E0. E0 makes impression. Z & E0 are linear actuators.

This is the sample gcode

```
T0                                                               // Select tool 1
G0 F5000                                                    // Set feed rate for X & Y movement 
G0 X10.0 Y10.0                                           // Move to position to make impression
G0 F500                                                      // Set feed rate
G1 E7                                                          // Make an impression of 7 mm
G1 E0                                                          // Return to 0
G0 F5000                                                    // Set feed rate for X & Y movement
G0 X111.0 Y57.0                                         // Place syringe over impression calculated based on the offset between Tool 0 and Z axis
G1 Z0                                                         // Lower X axis to 0
M42 P4 S255                                             // Turn GPIO pin ON
G4 P100                                                     //Delay 100 ms
M42 P4 S0                                                 // Turn GPIO pin OFF
G4 P500.0                                                 //Delay till dispensing finishes
G1 Z5.0                                                     // Raise Z axis and move to next position
```

The issue is GPIO pin turns on before reaching the position to dispense liquid.

Why is that?

I only want it to turn ON after `G1 Z0                                                         // Lower X axis to 0`

Is it possible??

Thanks

reloxx13, 2019-02-07:
maybe M400 - Finish Moves  ?

reloxx13, 2019-02-08:
close this if u r done, on fb u said it worked

athulsnair, 2019-02-09:
Sorry, So it performs as I want
Thanks

closed 2019-02-09 by athulsnair