MarlinFirmware/Marlin #13096: How to set Z endstop offset?

LinFor, 2019-02-07:
### Description

I want to disable all stuff like probing and ABL due to PROGMEM limitations. Now I have only an endstops and rigid printer construction, however, I still need features like "z-probe offset" and "babystepping" menus to quickly set compensation of hotend thermal expansion.
Yes, I have no probes, but I has dynamically changing Z-offset so I can't set it statically through the HOTEND_OFFSET_Z or so.

I find two possibilities:
1. Enable FIX_MOUNTED_PROBE and use available menu items "z-probe offset" and "z-offset babystepping" but I think this ideologically wrong. Also, using PROBE_MANUALLY what is ideologically right for me, hides "z-probe offset" and "z-offset babystepping" from menu.
2. Use "Set Home Offsets" menu item to shift whole workspace, but this wery unusable because I need only correct endstop compensation of real zero point of Z-axis (currently after M428 I keep seeing absolute coordinates of axes on status screen during print). Also, there are no LCD menu to manually edit M206 values, and without BABYSTEP_ZPROBE_OFFSET (unavailable without FIX_MOUNTED_PROBE) manual online correction not saved to EEPROM.

I think offset correction and relevant menu items should be available for all axis (maybe except extruder-axis) and for Z-axis in for Z-axis firstly regardless of the usages of probes.

LinFor, 2019-02-11:
@thinkyhead Please look at this. I can try to make a PR if you clarify concept of using this offsets to me.

gloomyandy, 2019-02-11:
What sort of printer and endstop setup do you have? With a conventional i3 style printer and microswitch Z min endstop then providing a mechanism that allows Z to be lowered below the endstop point will force the X gantry against the switch, this does not seem a very good idea.

LinFor, 2019-02-11:
I have a Flyingbear P905-based printer. It's a cube construction with Z-moving bed.
Yes, I can tune Z-endstop position physically, but it seems not as quick and easy as just babystep Z till bed reaches warmed nozzles.
I talking about 0.100-0.300mm correction values so I don't think that this could damage microswitch.

boelle, 2019-03-06:
@LinFor still working on this?

LinFor, 2019-03-12:
@boelle I do not work on this, because no one explained how conceptually it is supposed to work with axes boundaries.
It seems that the current behavior suits everyone, although it seems somewhat illogical to me.
At the moment, I have the FIX_MOUNTED_PROBE option enabled, although I don’t have a probe and the Z-axis limit is determined by endstop.

boelle, 2019-03-13:
and you tried these places?

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report.](https://github.com/MarlinFirmware/Marlin/issues/new/choose)

closed 2019-05-17 by thinkyhead