MarlinFirmware/Marlin #13101: z movement

electromc2, 2019-02-08:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

RudyBenner, 2019-02-08:
Did I have one too many refreshments?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-17 by thinkyhead