MarlinFirmware/Marlin #13102: TMC2130 with 3 differents models of motors on same print

tobecwb, 2019-02-08:
Currently, trying to use TMC2130 on a CR10 S-5.

- This printer has two pancake stepper motos on Z (I wired in series)
- No name NEMA 17 on X axis;
- A big stepper motor on Y axis (JD42HS60-1500A-02F -> https://3dprinting.stackexchange.com/a/7694)

I have TMC2130 on all axis (including E0)!
Currently, all motors are running at 24V, and almost everything works ok.

All TMC2130 has active cooling fan (a big fan of 200mm), and no prewarn triggered.
Current of all drivers are at default value (800ma, including Y)...

If I disable STEALTHCHOP, I got no layer shifts, but a horrible sound on Z motors... very loud!
If I enable STEALTHCHOP, silence... but layers shift **only Y axis**!

I already tried to increase the Y current until 1200, but no success... always got layer shift.

I believe something must be changed only for Y axis. This stepper motor is bigger  because move a bed of 500x500mm, but I really don't know where start.

PS: I already checked the belts, and every mechanical parts of Y axis.

landodragon141, 2019-02-08:
Try tuning the chopper settingd

teemuatlut, 2019-02-08:
Things to try
* Update to latest firmware
* Reduce Y acceleration and/or jerk
* Disable interpolation
* Enable junction deviation
* Set spreadCycle only for Y motor

It seems there's not much information about your motor, specifically the coil resistance and voltage rating.  There's no telling if your motor actually is a good choice in the first place.

tobecwb, 2019-02-08:
> Update to latest firmware
using Marlin 1.1.9 with latest TMC drivers

> Reduce Y acceleration and/or jerk
max acceleration was "1000", and jerk with 5.0

> Disable interpolation
with interpolation disabled, I got less layer shift, but still exists.

> Enable junction deviation
> Set spreadCycle only for Y motor
I will try

> Try tuning the chopper settingd
How? Any guide to indicate?

bruce356, 2019-02-08:
Just my opinion, but a 500x500 bed is quite large and heavy, slow right down to compensate. The TMC2130 drivers are probably not power full enough (they are a low power driver) for your application, try a different driver one that matches the amperage capacity of the motor on your Y axis, make sure the stepper is power full enough & fit for purpose.

tobecwb, 2019-02-09:
Yes, I was already thinking about changing the driver (at least for y axis).

InsanityAutomation, 2019-02-09:
Mostly agreed with Bruce above, but also try spreadcyle on em for better performance under load as a last resort.

boelle, 2019-03-06:
@tobecwb still working on this?

boelle, 2019-03-06:
and 2208 run cooler than 2130.

just remember to put fan on them so they get cooled