MarlinFirmware/Marlin #13103: Compile error with bugfix-2.0 when enabling Junction Deviation on a Delta configuration.

viper93458, 2019-02-08:
### Description

Compile error with bugfix-2.0 when enabling Junction Deviation on a Delta configuration:

sketch\src\module\configuration_store.cpp: In static member function 'static void MarlinSettings::reset()':

sketch\src\module\configuration_store.cpp:2039:32: error: 'DEFAULT_XJERK' was not declared in this scope
     planner.max_jerk[X_AXIS] = DEFAULT_XJERK;
                                ^
sketch\src\module\configuration_store.cpp:2040:32: error: 'DEFAULT_YJERK' was not declared in this scope
     planner.max_jerk[Y_AXIS] = DEFAULT_YJERK;
                                ^
sketch\src\module\configuration_store.cpp:2041:32: error: 'DEFAULT_ZJERK' was not declared in this scope
     planner.max_jerk[Z_AXIS] = DEFAULT_ZJERK;
                                ^
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

### Steps to Reproduce

1. Use the example config files for the FLSUN Delta - auto_calibrate (Only enabling junction_deviation)
2. Enable junction_deviation in configuration.h
3. Compile

**Expected behavior:** [What you expect to happen]

Compile would succeed.

**Actual behavior:** [What actually happens]
Compile error as shown above

#### Additional Information
This has been occurring for a little while (more than a month), however it's been quite a long time since I updated my Delta firmware with the latest 2.0 bugfix, so I have no idea when it actually started. I was using junction deviation previously (late last year) without a compile error. I started playing with current 2.0 bugfix versions within the last 30 days or so.

I tried searching the forums to see if anyone else reported similar issues, but I couldn't find anything.

As a side note (may or may not be related), with the current 2.0 bugfix tests I have done on the Delta (for the last month or so), when I start a print, the board resets during the first layer. If I turn the feedrate down to about 20 percent  it will run longer but still eventually resets. I do use octoprint to run the printer and wasn't having problems with bugfix 2.0 as of late last year. Again, I am not sure if its related, but I figured I would mention it. I can troubleshoot that more later on if desired and include my customized configs if needed.

I did see old forum posts about planner starvation and timeouts, but nothing recent.

Kind Regards,
William

Ludy87, 2019-02-08:
Search in `Marlin\src\inc\Conditionals_post.h` this line
`#define HAS_CLASSIC_JERK (IS_KINEMATIC || DISABLED(JUNCTION_DEVIATION))`
and replace this with
`#define HAS_CLASSIC_JERK (IS_KINEMATIC && DISABLED(JUNCTION_DEVIATION))`
and try again

viper93458, 2019-02-08:
I used your first option (that is now crossed out) and it worked to compile without error. I haven't tried your second option. Is one better than the other?

-William

viper93458, 2019-02-10:
Is one option better than the other? I will close this if it’s not actually a bug but intended to work this way. :)

Ludy87, 2019-02-10:
did not think the first solution was right.
maybe @thinkyhead can judge it better.

solrac8874, 2019-02-13:
I had the same compile error enabling Junction Deviation on my Delta.  Updating configuration_store.cpp per @Ludy87 option #2 compiled without error.  Not sure if that fixes it with Junction Deviation actually enabled.

boelle, 2019-03-06:
@viper93458 still working on this?

viper93458, 2019-03-06:
The solutions provided seem to work, but I don't know which is preferred. I was hoping that this would end up being fixed in the codebase, but thus far I haven't noticed it so I haven't closed this out.

-William

boelle, 2019-03-06:
at least you show interest in getting it solved, there are not many like you :-D

viper93458, 2019-03-06:
Certainly! I have enjoyed using Marlin and continually learning how the deeper features work. Unfortunately I lack the developer skills to contribute. :( 

Keep up the great work team!

closed 2019-05-17 by thinkyhead