MarlinFirmware/Marlin #13104: [2.0.x] Re-arm board heating bed 24V

jbaiao, 2019-02-08:
### Description

Auto PID too slow.
After done and changed in firmware, heating the heatbed the LED goes flashing and a very long time to reach a temperature.
It's all new and runs 24V.

Already tested with another power supply and does the same.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2845943/Configuration.zip)
https://youtu.be/lHkN6myhJSE

FanDjango, 2019-02-08:
How long does it take until 60°?

jbaiao, 2019-02-08:
About 8 minutes.

FanDjango, 2019-02-08:
40° in 8 minutes works out to 1° every 12 seconds. Is not bad, I think. But it should not blink at all below 50 (PID_FUNCTIONAL_RANGE = 10) unless probing and PROBE_HEATERS_OFF. 

I would expect a little better from 24V maybe.

Also, this needs to be adjusted, methinks, try that first:
```
  //120V 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
  //from FOPDT model - kp=.39 Tp=405 Tdead=66, Tc set to 79.2, aggressive factor of .15 (vs .1, 1, 10)
  #define DEFAULT_bedKp 160.33
  #define DEFAULT_bedKi 15.52
  #define DEFAULT_bedKd 414.06
```

My values are:
```
// FIND YOUR OWN: "M303 E-1 C8 S90" to run autotune on the bed at 90 degreesC for 8 cycles.
// Command M303 E-1 S65 C8 in OctoPrint Terminal §§§§
// Values from 20190208 §§§§
#define DEFAULT_bedKp 440.48 //§§§§
#define DEFAULT_bedKi 49.11  //§§§§
#define DEFAULT_bedKd 987.67 //§§§§
```

AnHardt, 2019-02-09:
40°C would only be right if you suppose the environments temperature to be at ~20°C. Here we have about 12°C or below (see extruder temp. at video). So heater power is even better.

Nerveless the questions remain:
Why is the bed-heater LED blinking?
Why that seems to be arhythmic? 

Defect PWM? Unlikely -  wouldn't we have more reports about that?
Pin conflict? Unlikely when using standard pin-file. Very likely when changed.
Defect or unsuitable or misused or misconnected hardware? Always possible.
Unexpected side effects of **recent** software changes?

Isn't standard RAMPS at 24V questionable? 
Fuses on RAMPS are always suspect. But 24V should reduce the current if the heater cartridge is matching - else that will probably be a problem.

AnHardt, 2019-02-09:
Sorry cited wrong issue. Too many bed problems. :-(

jbaiao, 2019-02-10:
Problem solved.
Every time I send full firmware, never clean de eeprom data. 
After write manual de PID value, everything works perfectly. 
There is anything that can be done to clean eeprom?

closed 2019-02-10 by jbaiao

FanDjango, 2019-02-10:
Initialize EEPROM, and/or LOAD FAILSAFE and then STORE CURRENT VALUES