MarlinFirmware/Marlin #13106: [Marlin-bugfix-2.0.x] MKS Gen 1.4 TMC2208 can't connect to UART

bakteriak, 2019-02-08:
### Description

Hi there, I am using Marlin 2.0, MKS Gen 1.4 board with 5 TMC2208 stepper drivers. I made Y connector (with resistor) which is used for the 5 steppers UART pins. I have followed this instruction .https://www.instructables.com/id/UART-This-Serial-Control-of-Stepper-Motors-With-th/

Pins
X-Axis TX 40 RX 63
Y-Axis TX 59 RX 64
Z-Axis TX 42 RX 65
E0 TX 44 RX 66
E1 TX 6 RX 11

Obviously I have modified pins to make sure they are correctly assigned.

If I run in stelthChop mode all drivers are working however steppers are moving double distance for some reason.

Unfortunately, I have a problem with connection to UART. Here is the output from M122 command. Appreciate if anyone can give me some pointers.

### Steps to Reproduce

22:42:50.048 : N19 M122*26
22:42:50.064 : X Y Z E
22:42:50.112 : Enabled false false false false
22:42:50.112 : Set current 800 800 800 800
22:42:50.128 : RMS current 1436 1436 1436 1436
22:42:50.171 : MAX current 2025 2025 2025 2025
22:42:50.171 : Run current 25/31 25/31 25/31 25/31
22:42:50.171 : Hold current 12/31 12/31 12/31 12/31
22:42:50.187 : CS actual 0/31 0/31 0/31 0/31
22:42:50.219 : PWM scale 0 0 0 0
22:42:50.235 : vsense 0=.325 0=.325 0=.325 0=.325
22:42:50.267 : stealthChop false false false false
22:42:50.283 : msteps 256 256 256 256
22:42:50.315 : tstep 0 0 0 0
22:42:50.315 : pwm
22:42:50.315 : threshold 0 00 0
22:42:50.315 : [mm/s]----
22:42:50.331 : OT prewarn false false false false
22:42:50.347 : OT prewarn has
22:42:50.347 : been triggered false false fals efalse
22:42:50.363 : off time 0 0 0 0
22:42:50.395 : blank time 16 16 16 16
22:42:50.395 : hysteresis
22:42:50.411 : -end-3-3-3-3
22:42:50.443 : -start 1 1 1 1
22:42:50.443 : Stallguard thrs
22:42:50.443 : DRVSTATUS X Y Z E
22:42:50.459 : stst
22:42:50.491 : olb
22:42:50.507 : ola
22:42:50.539 : s2gb
22:42:50.555 : s2ga
22:42:50.571 : otpw
22:42:50.603 : ot
22:42:50.620 : 157C
22:42:50.652 : 150C
22:42:50.668 : 143C
22:42:50.710 : 120C
22:42:50.710 : s2vsa
22:42:50.710 : s2vsb
22:42:50.710 : Driver registers:
22:42:50.710 : X0x00:00:00:00 Bad response!
22:42:50.710 : Y0x00:00:00:00 Bad response!
22:42:50.764 : Z0x00:00:00:00 Bad response!
22:42:50.764 : E0x00:00:00:00 Bad response!
22:42:50.764 : Testing X connection... Error: All LOW
22:42:50.764 : Testing Y connection... Error: All LOW
22:42:50.764 : Testing Z connection... Error: All LOW
22:42:50.764 : Testing E connection... Error: All LOW

Only E1 driver seems to be working. Other 4 doesn't move at all.

#### Additional Information

[Conf_files.zip](https://github.com/MarlinFirmware/Marlin/files/2847220/Conf_files.zip)

ErikFontanel, 2019-02-12:
Sounds like the same as #12800 unfortunately

boelle, 2019-03-06:
@bakteriak still working on this?

bakteriak, 2019-03-08:
@boelle this is still an issue

ncliver, 2019-03-29:
This is for anyone else who lands here looking to solve the double movement in Mode 0 issue: You need to modify DEFAULT_AXIS_STEPS_PER_UNIT the default is 80/80/400/93 I taped some graph paper to the bed and rubber banded a pen to the hotend and did a few 1 and 10mm movements then measured the lines on the paper to confirm it was actually moving exactly 2/20mm for me I changed the DEFAULT_AXIS_STEPS_PER_UNIT { 40, 40, 200, 47.5 } and everything worked fine.

closed 2019-05-17 by thinkyhead