MarlinFirmware/Marlin #13108: Fysetc f6 and Reprap discount full graphic smart controller sd card reader issue

stemaDev, 2019-02-09:
I've installed marlin on a Fysetc f6 board and a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER (connected with the EXT1 and EXT2), and everything seems to work fine, but it says "card inserted" when I put int the sd card but in the menu I see "no sd card", and it's empty if I try to open it. I took the pins file from fysetc's wiki page

### Steps to Reproduce

1. Use the pins_FYSETC_F6_13.h from fysetc's wiki page
2. define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and SDSUPPORT
3. flash fw
4. Connect the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER with EXT1 and EXT2
5. insert a card

**Expected behavior:** see the card's content and print from it

**Actual behavior:** "no sd card" text and empty

#### Additional Information

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/2847663/configs.zip)

ErikFontanel, 2019-02-12:
What happens if you send M021?

stemaDev, 2019-02-12:
I will test, but I have klipper installed now, because I want to print, and usb printing has bad quality because it cannot keep up.

Oalf72, 2019-02-15:
#REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
// IMPORTANT: The U8glib library is required for Graphical Display!
//            https://github.com/olikraus/U8glib_Arduino
plz check U8glib
AND
error in line 46 in the file ultralcd_impl_DOGM.h
was: #include <U8glib.h>
need to: #include "U8glib.h"
fix and try

Sprk-nl, 2019-03-10:
I'm in the same trouble and have some confusion which version of U8glib to use.
I have one in: user\Documents\ArduinoData\packages\fysetcf6\hardware\avr\1.3.0\libraries
Which is an old U8glib one: 6-4-2016
Marlin points in the Configuration.h to : https://github.com/olikraus/u8glib, 
and this website tells you this version is outdated and you have to use the U8G2, by downloading it via the Arduino Library Manager: https://github.com/olikraus/u8g2/wiki
Which U8glib library is the correct one?

boelle, 2019-03-10:
https://github.com/olikraus/u8glib

boelle, 2019-03-13:
@thinkyhead think we can assume the OP "dead" and close this one

closed 2019-05-17 by thinkyhead