MarlinFirmware/Marlin #13111: [Marlin-bugfix-2.0.x]  Difficult to explain

mr-miky, 2019-02-09:
Yesterday something strange happened to me.
I did a test print of the usual 3d boat.
The thing that leaves me puzzled and still can not explain is that up to 9mm in height everything seems a little atrophied. I compared it with another print made several months ago and I found the following:
(I state that the measurements were made with a caliber)

They are equal high so no layers have been lost and the Z axis does not lose any steps.
The base of the two boats is of equal width, so the x and y axes should have no problems.
After 9 mm in height, width and length are exact.
Below 9 mm there seems to be a constant proportional reduction factor on both the x and Y axes.

Linear advance is disabled, the eeprom has been reinitialized in order to avoid possible compatibility problems with previous firmware (always 2.0 of 3 weeks ago max).
I looked at all the Gcode layers used on octoprint and there is no undersize.

[image]

reloxx13, 2019-02-10:
do u use cura?

try another slicer

mr-miky, 2019-02-10:
I'm using Slicer prusa edition. Sliced gcode is correct.  

I've thought about it and I'm suspicious because the pla I used for the two tests is not the same. In addition to the firmware, the extrusion material has also changed. Probably the infill "pulls" progressively as it goes up and this would explain the deformation x, y and not on Z. When the infill ends everything seems to return to normal. I see if I can find some pla "old" type to do some further investigation.

closed 2019-02-10 by mr-miky