MarlinFirmware/Marlin #13112: Z_STEPPER_AUTO_ALIGN - compensate in wrong direction

gucio1200, 2019-02-09:
### Description

G28
G34 
Start measuring , and compensating in wrong direction - opposit

#### Configs
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2848044/Configuration.zip)

gucio1200, 2019-02-09:
Temp solution is to switch Z and Z2 cables.

AnHardt, 2019-02-09:
Swap either motors or endstops.

RudyBenner, 2019-02-11:
You have 2 Z steppers but are configured for 3.

Try this

/**
 * Z Steppers Auto-Alignment
 * Add the G34 command to align multiple Z steppers using a bed probe.
 */
#define Z_STEPPER_AUTO_ALIGN
#if ENABLED(Z_STEPPER_AUTO_ALIGN)
  // Define probe X and Y positions for Z1, Z2 [, Z3]
  #define Z_STEPPER_ALIGN_X { 25, 275 }
  #define Z_STEPPER_ALIGN_Y { 150, 150 }
  // Set number of iterations to align
  #define Z_STEPPER_ALIGN_ITERATIONS 3
  // Enable to restore leveling setup after operation
  #define RESTORE_LEVELING_AFTER_G34
  // Use the amplification factor to de-/increase correction step.
  // In case the stepper (spindle) position is further out than the test point
  // Use a value > 1. NOTE: This may cause instability
  #define Z_STEPPER_ALIGN_AMP 1.0
  // Stop criterion. If the accuracy is better than this stop iterating early
  #define Z_STEPPER_ALIGN_ACC 0.02
#endif

I have 12 iterations

bjbeare, 2019-02-23:
See: https://github.com/MarlinFirmware/Marlin/pull/13239

thinkyhead, 2019-02-23:
Please test the `bugfix-1.1.x` and/or `bugfix-2.0.x` branches to see whether it helps. If the problem has been resolved then we can close this issue. If the issue isn't resolved in either of those branches yet, then we should investigate further.

Minims, 2019-02-23:
Z Auto Align does compensate the probe position in bugfix-2.0
My Probe is X -34. I probe my first point in X+10. Currently the nozzle goes in X+10 instead of the Probe. It was OK with the version I was running 2 weeks ago. I've update to last commit, and this problem appears.

556duckvader, 2019-03-02:
I have downloaded the latest 2.0 version and it still appear to be not working.  I run G28 and G34 and it goes to my probing points and it still only moves my right side down each time.  It will not move the bed up at all.  I have even aligned the Z steppers by hand and it still moves the bed down each time I run G34 to the point it completely out of alignment.

gloomyandy, 2019-03-02:
Are you sure you just don't have the two Z axis the wrong way around? Have you tried swapping them over by simply exchanging the two Z cables? I've been using G34 with no problems with 2.0.x, I originally had the drives the wrong way around and had the same problem you are describing.

RudyBenner, 2019-03-02:
/**
 * Z Steppers Auto-Alignment
 * Add the G34 command to align multiple Z steppers using a bed probe.
 */
#define Z_STEPPER_AUTO_ALIGN
#if ENABLED(Z_STEPPER_AUTO_ALIGN)
  // Define probe X and Y positions for Z1, Z2 [, Z3]
  #define Z_STEPPER_ALIGN_X { **40, 180** }
  #define Z_STEPPER_ALIGN_Y { **110, 110** }
  // Set number of iterations to align
  #define Z_STEPPER_ALIGN_ITERATIONS **12**
  // Enable to restore leveling setup after operation
  #define RESTORE_LEVELING_AFTER_G34
  // Use the amplification factor to de-/increase correction step.
  // In case the stepper (spindle) position is further out than the test point
  // Use a value > 1. NOTE: This may cause instability
  #define Z_STEPPER_ALIGN_AMP 2   /1.6
  // Stop criterion. If the accuracy is better than this stop iterating early
  #define Z_STEPPER_ALIGN_ACC 0.02
#endif

AnHardt, 2019-03-03:
Maybe a negative `Z_STEPPER_ALIGN_AMP` could also do the job.

However. The used algorithm is probably not the best possible one.
Instead of a fixed factor one could calculate that from the difference between two sets of probings. Like in a 'Newton Approximation'. That is likely to converge much faster and would correct the factors sign automatically.

556duckvader, 2019-03-12:
> ```c++
> 2   /1.6
> ```

I applied this to my config and it still moves the right side down when I run the G34 command.  I have attached link to a video of what it is doing.
[https://www.youtube.com/watch?v=nsRCGrNgHiE](url)

556duckvader, 2019-03-12:
Just for fun I decided to swap the motors connections and it started working.  I am not sure what swapping motor connections did but it works now.

gucio1200, 2019-03-14:
Yup,

Aurawin, 2019-05-14:
After experiencing some problems with this issue I can say that with my dual z-axis wire orientation is verified (Z1->R (right), Z2->L (left)) I indeed (consistently) get a cumulative error.  

I used calipers to verify that the left axis was indeed misaligned.  I ran the alignment method and checked again but found the alignment method just added to the error making it twice as bad.  And infect the more I ran auto-align the more it lowered my L axis causing stress on the lead bushing.  Manually aligning the bed and running auto-alignment yields a very small error but it is again on the left side.  My guess is if I kept running it the compiled error would be in the MM.

If I swap at controller board Z1->Left, Z2-Right it appears to help but there is still something wonky - if you use calipers.  The swap results in a more level bed however, R was "lowered" compensate for the error that was on the left side instead of raising L at normal orientation.

I have disabled this feature for now.

RudyBenner, 2019-05-14:
Z1 is left, Z2 is right.

You did not include your config files.

Works just fine here.

Aurawin, 2019-05-14:
> Z1 is left, Z2 is right.
> 
> You did not include your config files.
> 
> Works just fine here.

Umm - that depends on your setup.  Z1 could be right or left.  LOL.

#define Z_STEPPER_AUTO_ALIGN
#if ENABLED(Z_STEPPER_AUTO_ALIGN)
  // Define probe X and Y positions for Z1, Z2 [, Z3]
  #define Z_STEPPER_ALIGN_X { 25, 248 }
  #define Z_STEPPER_ALIGN_Y { 150, 150 }

  // Set number of iterations to align
  #define Z_STEPPER_ALIGN_ITERATIONS 3
  // Enable to restore leveling setup after operation
  #define RESTORE_LEVELING_AFTER_G34
  // Use the amplification factor to de-/increase correction step.
  // In case the stepper (spindle) position is further out than the test point
  // Use a value > 1. NOTE: This may cause instability
  #define Z_STEPPER_ALIGN_AMP 1.0 // <<<---- tried tweaking this
  // Stop criterion. If the accuracy is better than this stop iterating early
  #define Z_STEPPER_ALIGN_ACC 0.02
#endif

Aurawin, 2019-05-14:
I was wondering - even with ABL, how DOES marlin know which stepper to lower and or raise?  With single z-axis it's indifferent.  With dual and triple Z it would be an entirely different matter.

Which brings me to the thought that there may be some assumptions made with Marlin to leveling and adjusting of axis and or build-pate.

Maybe there should be a section inside the configuration.h that clearly requires attention by end-users to provide marlin with the layout to which stepper is at X,Y(R)  and  X,Y(L)

This may effect ABL and Z-align

boelle, 2019-06-22:
missing label

sjasonsmith, 2019-06-26:
I am seeing the same issue on a printer with 3 Z motors. I am quite sure that I have verified that the order of the probe points matches the motor order. I will double-check this tomorrow to remove any doubt.

Unlike a 2 motor machine, I cannot reverse the probe order, because "reversing" a correction would require moving two other motors, not just one.

I have modified the G34 code to invert the correction direction, and I have successfully auto-aligned my bed with this change.

I can create a pull request to fix this, but it will break everybody who has worked around this for a 2-motor machine. I doubt any 3-motor users have been using this successfully, but I would love for somebody to prove me wrong!

Does anybody have a suggestion for how to fix this without breaking existing users, or to at least break them at compile time and not when they crash their print heads?

AnHardt, 2019-06-26:
> or to at least break them at compile time and not when they crash their print heads?

Option renaming. Sanity check for the old name giving instructions.

AnHardt, 2019-06-26:
The algorithm is crap anyway.
Try a Newton approximation. Measure n_axis points, move the spindel with the largest error by a small amount (for example half the error), \* measure again, calculate the change of differences and how much to go in next round to be right (approximating by assuming linearity). Move the spindel with the highes error by the calculated way. Repeat from \* until exact enough or not improving anymore.

sjasonsmith, 2019-07-07:
> I am quite sure that I have verified that the order of the probe points matches the motor order. I will double-check this tomorrow to remove any doubt.

I checked again and actually did have my motors in the incorrect order. The wiring in this printer is hard to access, and my initial method of verifying the order was flawed. With the order fixed I am seeing moves in the correct direction with the current code.

RudyBenner, 2019-07-07:
Excellent.

sjasonsmith, 2019-09-21:
I have spent a lot of time with Z_STEPPER_AUTO_ALIGN in my work to improve my triple-Z behavior. This includes testing with my printer only using two Z motors.

I am of the opinion that if this particular bug was an issue, it has since been fixed, and this bug should be closed.

InsanityAutomation, 2019-09-21:
I agree, every case of this I have seen was steppers reversed from expected ordering. I intend to make this configurable at some point.

boelle, 2019-10-12:
@gucio1200 is this still a problem for you?

InsanityAutomation, 2019-10-15:
@boelle No, this should be closed

closed 2019-10-15 by boelle

aaronmrosenthal, 2020-03-11:
I have been having issues with G34 3 steppers...

1. I don't know if the order is proper. 2 in the back, 1 in the front.
Core XY
Order:
2-1
-3- 

Seems G34 deviates the wrong way 

I set 3 stepper, orientation to 2, just want to be sure I am doing this right. 

I am also having issues probing M422 S3 X250 Y20 always giving invalid probe area even though the bed area is 400x550

Thanks for any help with this! Trying to manually level 400x550 belts is bloody hell.

thinkyhead, 2020-03-11:
@aaronmrosenthal — We just merged a `G34` patch today. please give the latest a try.

We don't do any kind of support to help out end users here. So if you need additional help with configuration please take advantage of these helpful resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y