MarlinFirmware/Marlin #13114: [bugfix 2.0] Button not working for pause on stow/deploy with G29 P1

pandel, 2019-02-09:
Printer: Anycubic Kossel Linear Plus with Trigorilla 1.4, UBL active

PAUSE_BEFORE_DEPLOY_STOW is defined / activated in Configuration.h

When I run the step one from UBL mesh generation menu, the printer first executes a G28 and then a G29 P1. During this it asks me to "Deploy Z-Probe", but doesn't react when I hit the button. It simply waits forever.

BUT: when I run "Delta Auto Calibration" it works as expected. The printer is homing all axis, drops the nozzle a bit, asks for "Deploy Z-Probe", I attach the probe and hit the button and the printer begins calibration.

What's going wrong here?

pixelbucket, 2019-02-26:
Sending M108 will get you past "Deploy Z-Probe" but unfortunately you'll hit the same problem at "Stow Z-Probe" and the same solution won't work a second time.

boelle, 2019-03-06:
tried with todays commit of 2.0?

boelle, 2019-03-06:
@pandel oki, maybe click close below and open new issue in future if you find the problem again?

pandel, 2019-03-06:
@boelle Ok, sure :-)!

closed 2019-03-06 by pandel

robbycandra, 2019-04-15:
Try with bugfix2.0 the problem still exist.