MarlinFirmware/Marlin #13115: [bugfix 2.0] RE-ARM +Reprap discount full graphic smart controller ENCODER PROBLEM

KuterTm, 2019-02-10:
Hello
I have problem with encoder on smart controler, Encoder is 100% good becouse i test it on mega than on ramps, and on ramps its "jumping" I turn in one direction, but marlin think its is turning back and forrword.

Behavior is like on movie in You tube.
First is try on Mega, second try is on RE-ARM
https://www.youtube.com/watch?v=mFQCTlHibHQ

1. Try encoder settings from marlin 1.1.X
2. change encode seting so smaller - biger
...

**Expected behavior:** [What you expect to happen]
It will be work good :) 

ZIP with configs in attachment

mr-miky, 2019-02-10:
It would seem an electrical problem rather than software. It seems that there is a "bouncing" between the two quadrature signals of the ecnoder maybe the pullup on the inputs dedicated to the encoder is missing or the encoder signals are dirty.
Check that there is a good connection between the display / encoder GND and the ramps

KuterTm, 2019-02-10:
Sorry but You are wrong. If You watch youtube movie i first use Arduino mega board and there wasent any  problem than in fiew second i plug ramps to RE-arm board and there was a problem.  But to be shure i change after that All LCD, conectors to LCD (tapes) i change board to second ramps 1.4 and than i use ramps 1.6 plus, on all is this problem so sorry i change what i can and encoder is still bouncing

KuterTm, 2019-02-11:
Hello
For test i just now instal difriend firmware - Smoothie, and there was no problem with encoder :( i back to marlin becouse its better for me, but encoder is crazy

J-charles-C, 2019-02-12:
I,

I have Re-arm + Ramps + Reprap discount full graphic smart controller and all is ok. 

Have you Test another commit of Marlin 2.0 ?

best

boelle, 2019-03-06:
i also use 2.0 and re-arm and full graph display, no issue

boelle, 2019-03-13:
@thinkyhead think we can close this one too

closed 2019-05-17 by thinkyhead

pocker, 2019-05-23:
Hi
I have the same problem. I had MKS Base V1.4 (came with tevo tarantula). I installed BIGTREETECH-SKR-V1.3 and with the new board my encoder jumping like KuterTm's but if i try with the old board it's working fine.

Nethersoul, 2019-05-23:
I observe the same effect with the ReprapDiscount smart controller and the Ulticontroller in combination with a 1.5.7 umo. I can load the original firmware and it works but with 1.1.x it jumps.
I am not latest and I have some TMC2208 hacked over uart which might have impact..

pocker, 2019-05-24:
To detect the encoder direction the marlin uses two pin. [Here is an example](https://howtomechatronics.com/tutorials/arduino/rotary-encoder-works-use-arduino/). I flashed the same marlin 2 to my old board and the encoder worked well so I think its a hardware problem(e.g. missing pullup resistor..etc) or wrong board configuration.