MarlinFirmware/Marlin #13117: Logic error in UMB

drpjk, 2019-02-10:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

drpjk, 2019-02-10:
Mea culpa. It must be too late at night. the logic is correct. Please delete post

gloomyandy, 2019-02-10:
If this is no longer an issue, please close it.

Roxy-3D, 2019-02-11:
> Mea culpa. It must be too late at night. the logic is correct. Please delete post

Not a problem!  It is nice when a problem just goes away.   :)

closed 2019-02-11 by Roxy-3D