MarlinFirmware/Marlin #13118: The blue TEMP_STAT_LED does not turn on after startup

Santobert, 2019-02-10:
### Description

I have installed two TEMP_STAT_LEDs onto a MKS Gen L. I've changed my configuration.h to:
```
#define TEMP_STAT_LEDS
#if ENABLED(TEMP_STAT_LEDS)
  #define STAT_LED_RED_PIN 6
  #define STAT_LED_BLUE_PIN 11
#endif
```

When the printer heats up, the red LED lights up as expected. After cooling down, the blue LED turns on.
But immediately after startup, no LED is on until the printer heats up.

### Steps to Reproduce

1. Change configuration.h as described
2. Flash firmware
3. Turn on the printer (no led is on)
4. Heat up (red is on)
5. Cool down and wait (blue is on)

**Expected behavior:** The blue led turns on immediately after startup

**Actual behavior:** No led is on

GMagician, 2019-02-10:
Fixed

Santobert, 2019-02-10:
Wow that was fast :)

When will it be merged into branch 1.1.x ?

GMagician, 2019-02-10:
I suspect 1.1.x branch doesn't accept PR anymore (this is for 2.0), but ready to post one if allowed

Santobert, 2019-02-10:
Oh sad. Maybe i should upgrade to bugfix-2.0.x.
Actually i wanted to wait for the release of stable 2.0.x

GMagician, 2019-02-10:
I'm using 2.0.x with mega2560 and it works like a charms

Santobert, 2019-02-10:
I'll give it a try. Thank you

closed 2019-02-10 by Santobert

GMagician, 2019-02-10:
@Santobert I also posted 1.1.x version...even if not merged you may just get source

PR #13121 (2019-02-11): [1.1.x] Fix TEMP_STAT_LED startup init
Blue led is not turned on startup if it should (Fix #13118)
PR #13120 (2019-02-11): [2.0.x] Fix TEMP_STAT_LED startup init
Blue led is not turned on startup if it should (Fix #13118)