MarlinFirmware/Marlin #13126: can´t select negative zoffset

hugolainch, 2019-02-11:
Hi all!
I have adapted an E3D extruder on my Atom 2.0 (delta printer) to have the possibility of using an all metal extruder. I am using an external zprobe with has the following offset x=0 y=-16.90 z=-5.0.
I need some help with the firm (based on v1.0.0) because I have updated the values and the G29 auto leveling works fine and then the firm applies the following fórmula: zoffset=zprobe_offset+LCDzoffset_selected.
My problem is that my zprobe has an offset of -4.5mm with the nozzle so, when the firm applies last formula I get a negative value for zoffset, and when I tried to adjust for the first layer the lcd puts extrange symbols (see attached). 
It is caused because you cant select a negative value for the zoffset, so to adjust the offset I have to select a possitive value for the offset and repeat again the bed leveling to obtain the new negative value.
After 6 iterations I had been able to adjust fine but i have to repeat the bed leveling each time that i modify the zoffset that is crazy.
I have found the line where It is supposed you select the zoffset on lcd. "Ultralcd.cpp" As you can see the range is between -10 and 70. I can reach 70 on the lcd but I cant reach less than 0...

configuration.h
#define Z_PROBE_OFFSET {0, -16.9, -4.84, 0} // X, Y, Z, E distance between hotend nozzle and deployed bed leveling probe.

marlin_main.cpp
deploy_z_probe();

z_offset = z_probe_offset[Z_AXIS] + (code_seen(axis_codes[Z_AXIS]) ? code_value() : z_offset);
calibrate_print_surface();
retract_z_probe();

feedrate = 6000;
feedmultiply = saved_feedmultiply;
previous_millis_cmd = millis();
endstops_hit_on_purpose();

Config_StoreSettings();
break;
case 30: // G29 Calibrate print surface with automatic Z probe.
saved_feedrate = 6000*3.5;
saved_feedmultiply = feedmultiply;
feedmultiply = 100;

home_delta_axis(); //Bato add for auto homing ATOM when G29

ultralcd.cpp
static void lcd_setting_zoffset_menu()
{
START_MENU();
MENU_ITEM(back, MSG_MAIN, lcd_main_menu);
MENU_ITEM(gcode, MSG_AUTO_LEVEL, PSTR("G29"));
MENU_ITEM(function, MSG_GOTO_ZERO, deploy_z_zero);
MENU_ITEM_EDIT_CALLBACK(float52, MSG_ZOFFSET, &z_offset, -10.0, 70, Config_StoreSettings);
END_MENU();
}

[ATOM-2.0.6- Bondtech 200x830.zip](https://github.com/MarlinFirmware/Marlin/files/2849588/ATOM-2.0.6-.Bondtech.200x830.zip)
[image]
[image]
[image]

boelle, 2019-02-11:
my config looks like this:

#define X_PROBE_OFFSET_FROM_EXTRUDER 23  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 9  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.26   // Z offset: -below +above  [the nozzle]

hugolainch, 2019-02-11:
Thanks for your reply!
And with that configuration are you able to select a negative offset on the lcd screen?

boelle, 2019-02-11:
first of all the lines from your config is VERY different

what version of marlin do you run?

boelle, 2019-02-11:
will check about LCD when my print is done, but i never change it

hugolainch, 2019-02-11:
It is the firm suministrated from the Atom 3dp printer Seller. I think is based on v1.0.0.
The zprobe offsets are a bit different defined on  configuration.h file and It works, but when i try to regulate It ok the lcd It crashes for negative zoffset values. However as you can see on ultralcd.cpp It is defined a range between -10 and 70!

boelle, 2019-02-11:
i would start with making sure what you got

maybe upgrade to MArlin 2.0

without knowing precise what you got its hards to diagnose the problem

hugolainch, 2019-02-11:
/* -*- c++ -*- */

/*
    Reprap firmware based on Sprinter and grbl.
 Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 
 This program is free software: you can redistribute it and/or modify
 it under the terms of the GNU General Public License as published by
 the Free Software Foundation, either version 3 of the License, or
 (at your option) any later version.
 
 This program is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 GNU General Public License for more details.
 
 You should have received a copy of the GNU General Public License
 along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */

/*
 This firmware is a mashup between Sprinter and grbl.
  (https://github.com/kliment/Sprinter)
  (https://github.com/simen/grbl/tree)
 
 It has preliminary support for Matthew Roberts advance algorithm 
    http://reprap.org/pipermail/reprap-dev/2011-May/003323.html
 */

/* All the implementation is done in *.cpp files to get better compatibility with avr-gcc without the Arduino IDE */
/* Use this file to help the Arduino IDE find which Arduino libraries are needed and to keep documentation on GCode */

#include "Configuration.h"
#include "pins.h"

#ifdef ULTRA_LCD
  #if defined(LCD_I2C_TYPE_PCF8575)
    #include <Wire.h>
    #include <LiquidCrystal_I2C.h>
  #elif defined(LCD_I2C_TYPE_MCP23017) || defined(LCD_I2C_TYPE_MCP23008)
    #include <Wire.h>
    #include <LiquidTWI2.h>
  #elif defined(DOGLCD)
    #include <U8glib.h> // library for graphics LCD by Oli Kraus (https://code.google.com/p/u8glib/)
  #else
    #include <LiquidCrystal.h> // library for character LCD
  #endif
#endif

#if defined(DIGIPOTSS_PIN) && DIGIPOTSS_PIN > -1
#include <SPI.h>
#endif

boelle, 2019-03-06:
so you upgraded to bugfix 2.0?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-05-17 by thinkyhead