MarlinFirmware/Marlin #13127: 1 Extruder +

ROSnoob, 2019-02-11:
Hi , 

I have an Extruder MK8 and a card Ramps 1.4 installed on a arduino mega 2560, I would like to do 3D printing tests, to be able to install all the systems on a Kuka robot, I realized all the connections, but on arduino and precisely in the configruation file.h I selected the EFF mode because I do not have the Bed or external motors, my question is how can we extrude it directly on arduino, and how can we ignore all configurations because i'm not concerned about (engines, the bed , stepper)

Ps: this is the first time I use the ramps, I do not have much knowledge on the subject, thank you for your help and your attention

AnHardt, 2019-02-11:
(Currently) impossible.
Extrusion needs to be synchronized with the xyz-motion., so should be managed by the same controller.
The kuka probably can't handle the e-stepper (likely has servo motors)
Marlin currently can't handle the kukas sevo-motors (completely different from model-servos).
Sending the same g-code to both controllers at the same time will not synchronize tight enough.
An other way to synchronize is currently not available.

boelle, 2019-03-13:
@ROSnoob still working on this?

ROSnoob, 2019-03-14:
@boelle Yes , I've solved this by creating my own Arduino program without using the entire firmware, I just took the functions that activates the extruder and the stepper motor that has the role of pushing the plastic wire, and for movements I just programmed my robot independently of Marlin .

I close issue, thanks all

boelle, 2019-03-14:
remember to click close below :-D

closed 2019-03-14 by ROSnoob

EdisonAltamirano, 2020-05-20:
@ROSnoob  could you please share your code of the Arduino extruder communication with Kuka?