MarlinFirmware/Marlin #13128: Marlin 1.1.9 - BLTouch random trigger

italocjs, 2019-02-11:
I'm pretty sure that this is not a software issue, but i ran out of places to seek help, and maybe someone had this before.  I'm guessing my problem is electromagnetic interference

The problem is that the bltouch works as expected during homing and auto-leveling, but sometimes during the printer the probe goes down and up.  usually its really fast (less than 0,5 second).  but depending on the place this happens, it makes the printer lose steps and damage the pin.

A few days ago i managed to solve another issue where sometimes the marlin simply reports that the printer has stopped ( [ERROR] Error:Printer halted. kill() called!) when homing,  i managed to fix it installing a 10nf capacitor on the endstops, but it didnt worked in the bltouch.

Did anyone had this issue? and if yes, how was is solved?

Also, i'm using a awg24 flat cable, with 25ways to power two steppers (x and E0), the nozzle, fans and bltouch. all connections are perfect.

FanDjango, 2019-02-11:
Have you got the 5V for the BLTouch going on a single strand of the awg24 flat cable? 
I am doing more or less the same: 15 strands awg24. Got 3 + 3 for the nozzle, 4 for E stepper, bltouch, all single, no fans (controller board test printer). In my case it works.
Ideas:
Space multiple GND strands around the the BLTouch signal line (if it is a flat cable). Keep stepper away from BLTouch signal line as far as possible.
Use multiple strands for GND and +5V of the BLTouch if you can, and make sure they are not the same as the minus line of the nozzle - it is NOT GND and not minus depending on the circuit of the heater driver.
Put a 10uF low ESR capacitor on the 5V against the BLTouch GND line near the BLTouch to mitigate voltage drops on the supply line.
You cannot really denoise the BLTouch signal line as well as you did with the endstops but the idea was not bad.
If the needle drops down: another possible issue is that on BLTouch clones, vibrations can cause the needle to drop, I have read that in some forums. But only on really bad clones.

italocjs, 2019-02-11:
Hi FanDjango, thanks for helping.

There is a single 5V wire which is only used by the bltouch in the printing head. Everything else has separated power (either 15v or 10v).  I'm not so sure about this minus line, i'm getting the ground, 5v, 10v and 15v directly from the PSU, One thing you said that made me curious, is to measure if there actually is 5V in the bltouch, maybe some voltage was lost in the cable (about 80cm) and its causing the issue. 

I already tried replacing the bltouch with two new ones, no change.

I'll be trying now this capacitor, i should place it between the GND and 5V right? i just read something similar about servos, where a guy used a 10nf between signal and gnd too,

FanDjango, 2019-02-11:
GND at the controller board might be some millivolts higher than GND at the PSU due to current coming back from the controller. It all depends on whether you have separate mosfets or not and where their GND is connected. Best of course is: All GNDs meet at the PSU. But the BLTouch GND line should be connected to the controller board, separately, so that the BLTouch signal line is not offset by any spikes on your common high power GND line. Measurement with a analog voltmeter is no good: It averages any quick voltage spikes. Using a digital multimeter is also no good - it might average, it might give you fluctuating readings as momentary peeks into a moving target. To see if there are bad (microsecond) voltage drops, you need an oscilloscope (a good one, actually).
Just to be safe: once again beware that neither one of the two nozzle power lines (typically) are GND. It's typically +Vcc switched by open-drain mosfet down to GND at the location of the mosfet - either on the controller board or a separate mosfet. Unless you have a completely different wiring setup :-)
Sounds to me like you have a single small GND wire going to the PSU.
I would use multiple (say 3) GND lines to wherever you have them going. To be safe, give the Nozzle + Line more than one wire. Nozzle on my system uses 3A at 12V, 6A at 24V. Google the carrying capacity of single line awg24 - it is 3.5 amps max [https://www.powerstream.com/Wire_Size.htm](wire sizesl)
I would give the BLTouch a separate GND (better call it +0V) line to the controller.
And the capacitor goes over the GND and 5V right directly as close to the BLTouch as possible. But it won't perform miracles if the general setup has voltage drops. Since the 5V comes from the controller board, it should be ok, but the GND might be flaky. The cap won't fix that.

italocjs, 2019-02-11:
Just added a 10nf cap between gnd and "s",  i'll start printing now to see if the issue still happens.

[image]

FanDjango, 2019-02-11:
10nf might help. I meant more like a small (SMD, even) 10uF actually.

Can you make a sort of sketch of all your carriage relates connections? 

> maybe some voltage was lost in the cable (about 80cm)
Remember, you only lose voltage on the 5V if there is significant current being pulled on it. But that current which is going BACK on the GND will RAISE the GND voltage at the print carriage above the GND voltage at the controller (and PSU), thereby making a signal of the BLTouch look lower (smaller) at the controller than it really is when measured that the carriage.

FanDjango, 2019-02-11:
Oh, and that's a clone, looks like it. They actuall pull significant current on the 5V line, but only when they pull the needle up, I thought. I can actually see the LCD display-lighting flicker when they probe, which tells me they are not as good as the original BLTouch on my other printer (same setup - no flickering LCD light). It also tells me that the controller is not giving the best 5V, because the display also uses it.
What is your controller?

FanDjango, 2019-02-11:
One more thing: Of course it could also still be electromagnetic interference (crosstalk) on your flat cable - you can prove that by temporarily (for testing only) cabling the BLTouch with a separate really super (maybe even shielded) separate cable to the controller -just to see if that proves EMI is the problem.

italocjs, 2019-02-11:
i'm not using the arduino 5v (from the cheap 7805 thingy),  i'm using a 3amp stepdown to power both the arduino, the raspberry and the bltouch.  (i measured and the output current is at about 0,8amp, with the raspberry and lcd on)

This is my wiring

Flat | Wire pos | Ramps pin | Description | Wire color | Description
-- | -- | -- | -- | -- | --
FLAT A - MOTORES
A | 1 | Driver E0 A1 | Motor E0 A1 |   | Extruder   motor
A | 2 | Driver E0 A2 | Motor E0 A2 |  
A | 3 | Driver E0 B1 | Motor E0 B1 |  
A | 4 | Driver E0 B2 | Motor E0 B2 |  
A | 5 | Driver X A1 | Motor X A1 |   | X motor
A | 6 | Driver X A2 | Motor X A2 |  
A | 7 | Driver X B1 | Motor X B1 |  
A | 8 | Driver X B2 | Motor X B2 |  
A | 9 |   |   |   |  
A | 10 |   |   |  
FLAT B - SENSORES
B | 1 | X MAX - | Endstop X |   | Endstop X
B | 2 | X MAX S | Endstop X |  
B | 3 | SERVO - | BLTOUCH - | brown and black | BL TOUCH
B | 4 | SERVO + | BLTOUCH + | red
B | 5 | SERVO S | BLTOUCH PWM | Orange
B | 6 | Z MIN S | BLTOUCH S | white
B | 7 | T0 | thermistor | white | nozzle   thermistor
B | 8 | T0 | thermistor | white
B | 9 |   |   |   |  
B | 10 |   |   |   |  
FLAT 3 - BICO
C | 1 | GND Fonte | fan- | black | heatsink fan
C | 2 | 10V LM2596 | fan+ | red
C | 3 | D9 - | fan- | black | printing fan
C | 4 | D9 + | fan+ | red
C | 5 | D10 - | nozzle - | Orange | 40W heater
C | 6 | D10 - | nozzle - | Orange
C | 7 | D10 - | nozzle - | Orange
C | 8 | D10 + | nozzle + | Orange
C | 9 | D10 + | nozzle + | Orange
C | 10 | D10 + | nozzle + | Orange

FanDjango, 2019-02-11:
Are the three flat cables lying flat on top of each other?

FanDjango, 2019-02-11:
Maybe for testing put B6 on a totally separate single cable

italocjs, 2019-02-11:
yep! it was visually better to have three cables one on top of the other, here is a pic of what it looks like.  The machine is printing now, i'll wait to see if there is any issue, if there is i'll try passing another shielded cable to the bltouch

[image]

FanDjango, 2019-02-11:
Ok, I can see your Nozzel GND is separate. Disregard some of the prvious posts. You should have posted the cabling in the initial post. 
As matter of interest: Where do B3 and C1 (via which intermediate stations) meet up together?

FanDjango, 2019-02-11:
Anyway, B3 should go to the controller, and for EMI-safe signal paths you should get the BLTouch 5V from the 7805 on the controller board. Keep the BLTouch on it own.

FanDjango, 2019-02-11:
I like the cable.

FanDjango, 2019-02-11:
So B6 is sandwiched betweem A6 (stepper) and C6 (nozzle -, which is being PWMed). That's not a peaceful place to be.
For now I cannot help you beyond what we have discussed in the previous posts. But I am curious to hear how you progress, please keep us up to date. It is exactly these kind of things that too often get in the way when putting together a "beautiful" machine and others can certainly profit from this.

italocjs, 2019-02-11:
B3 and C1 meet in the PSU negative pin.

Until now i didnt noticed the bltouch going down (the only thing i did was to install a 10nf cap, which i took from a uln2003 board), i'll keep monitoring it over the day and post a feedback.  the issue was happening every 2-3 min.

i'll keep your recommendations in mind and modify my cable pinout, Thanks for all the help!

boelle, 2019-03-06:
@italocjs still working on this?

boelle, 2019-03-13:
@thinkyhead think we can close this one, OP is "dead"

italocjs, 2019-03-13:
Hey, for some reason i didnt received any notifications about your previous message boelle,  I managed to make it works adding the 10nf capacitor in the bltouch as posted on the image.   Im closing it now!

closed 2019-03-13 by italocjs