MarlinFirmware/Marlin #13130: [RFC] Virtual tools

revilor, 2019-02-12:
I'm working on an IDEX build at the moment and would like to attach a MMU2 to one of the extruders or even build a second MMU2 and have one for each of the extruders. At the moment there is no option in Marlin to configure this kind of setup. 

Or think of an E3D toolchanger-style printer where you attach a MMU2 to one of the toolheads and have another tool use a 2/3/5/x-color mixing extruder. I may be wrong but I can see no way how one could configure this kind of hardware setup in the way extruders/tools are defined and handled in Marlin at the moment.

Therefore I'm thinking of virtualizing tools on top of real hardware (extruders, hotends, fans, filament feeders, extruder multiplexers, filament sensors, ...) the same way computers can be virtualized on top of real CPU cores, RAM, hard disks, etc. 

So on my IDEX I have 

 - 2 extruders E0, E1
- 2 hotends H0, H1
- 2 part cooling fans FAN0, FAN1
- 1 MMU2 with 5 filament lanes MMU0 .. MMU4

On top of that I can define 6 virtual tools

T0 (E0, H0, MMU0, FAN0)
T1 (E0, H0, MMU1, FAN0)
T2 (E0, H0, MMU2, FAN0)
T3 (E0, H0, MMU3, FAN0)
T4 (E0, H0, MMU4, FAN0)
T5 (E1, H1, FAN1)

For the toolchanger example above with 4 toolheads,  one MMU2 and a 2-color mixing extruder there would be

 - 5 extruders E0 .. E4 (E3, E4 for the mixer)
- 4 hotends H0 .. H3
- 4 part cooling fans FAN0 .. FAN3
- 1 MMU2 with 5 filament lanes MMU0 .. MMU4

Those parts give us 8 tools

T0 (E0, H0, MMU0, FAN0)
T1 (E0, H0, MMU1, FAN0)
T2 (E0, H0, MMU2, FAN0)
T3 (E0, H0, MMU3, FAN0)
T4 (E0, H0, MMU4, FAN0)
T5 (E1, H1, FAN1)
T6 (E2, H2, FAN2)
T7 (MIX(E3, E4), H3, FAN3)

It gets more complex when you include different thermistors for the hotends or filament sensors, so I'll stop here. I guess you get the idea anyway.

Because this would be a huge project with potential implications throughout all of the Marlin code I would like to collect your thoughts, comments, ideas on this topic. I'm committed to work on this but probably would also need help to do so. Therefore I want to check first if this a way the Marlin community would like to go.

Thank you for listening!

InsanityAutomation, 2019-02-17:
Personally, I like the idea. It allows alot of flexibility for IDEX and multi tool systems in general.

Ill pull in @Roxy-3D as most of the idex code lately has been between the two of us.

Roxy-3D, 2019-02-19:
Actually....   I've been thinking about this topic.    I was toying with the idea of approaching a few of the IDEX vendors and seeing if any of them were interested in doing a product like this.     If they could do the hardware design to (initially) get a 3 color mixing extruder on both carriages, I can coordinate getting the firmware done for them.    

It may make sense to start a thread with that as the topic and see if any IDEX vendor wants to enter that space.   Because no hardware currently exists to do this, it isn't reasonable to expect or demand 'Production Level' equipment to do the development.   The next best thing would be some 'very nice prototype' hardware.     It would only cost the vendor a few good quality prototypes and in 6 months they would have the firmware to be shipping product.

Roxy-3D, 2019-03-05:
No interest from any of the companies that manufacture IDEX machines....

Oh well...    I bet there are Delta companies that would like to do this....

boelle, 2019-10-24:
would it be wrong to call this a feature request instead?