MarlinFirmware/Marlin #13131: [FR] Temperature sensor calibration

Deneteus, 2019-02-12:
Feature Request: Temperature Sensor calibration and Troubleshooting Documentation

Marlin needs a feature where you can take a known calibrated temperature sensor/thermocouple and use it to calibrate a thermistor to get the correct temps. You could also use this info to assist PID Autotune in coming up with better values. 

I say this as well because I don't happen to see a well documented way of making sure that the thermistors that we get are accurate. We just assume they are based on known examples but people buying them off eBay or Amazon have no idea if what they are getting is working correctly. 

- Ex1: There is an Ender-3 user saying that his temps are off by 20c when tested using a K type vs the stock when measured from within the hot end area.

[image]

[image]

- Ex2: A video showing this issue:  https://youtu.be/brl1weuD8bs

Note: I am aware of the following reasons that the temp could be off from the expect values. (PID Tune bounce not withstanding).

- Thermistor is faulty. (Does not meet Tolerance or mechanically defective,)
- Incorrect Thermistor Value (TEMP_SENSOR_0 configured for the wrong table Value.)
- The user or vendor purchased the incorrect replacement thermistor for their stock firmware.
- The user or vendor has the incorrect thermistor, and their firmware (TEMP_SENSOR) is incorrect.
- The users 3D Printer board thermistor inputs could be faulty. 

**Expectation:**  

- Marlin documentation should have a well documented troubleshooting process for when this issue arises. How do we know when the thermistor or printer board thermistor input is operating correctly.

- Marlin 2.0 should be able to create its own thermistor table and save it to SD or output it as text. 

- Marlin 2.0 should be able to tell you when a thermistor value may be wrong or show an error.

Sincerely,

Derrick Smith

AnHardt, 2019-02-12:
> ...
> Marlin needs a feature where you can take a known calibrated temperature sensor/thermocouple and use it to calibrate a thermistor to get the correct temps. ...

It already has.

> ... You could also use this info to assist PID Autotune in coming up with better values.

Autotune does its job based on the temperature readings it gets. It can't judge how correct this readings are. For the the same (correct) target temperature (and the same environment conditions and same 'system') it produces fairly repeatable results when done with a high amount of cycles. Only it you'd know the 'correct' result and get an other you'd know, something went wrong. But if you don't expect 'new' PID values you'd not start autotune.

> I say this as well because I don't happen to see a well documented way of making sure that the thermistors that we get are accurate. 

There are a few indicators.
a) Nozzle- and bed- temperature are not fairly the same when at room temperature for a long time.
b) Temperature at room temperature is not about the same as your room thermometer.
c) Heating is unusually slow or fast.
d) Filament with known melting temperature is melting far above/below measured temperature.

Non of this can be done automatically. It's in your responsibility the check that.

> We just assume they are based on known examples but people buying them off eBay or Amazon have no idea if what they are getting is working correctly.

You are right. Some users are naive. But most of them wouldn't order new tires for their cars without knowing the exact size, but order a thermistor without having any further data. (I guess they spent money and get experience for that - not a bad deal - even if the part is completely unusable - or - especially then - compared tho the tires its a cheep experience.)

> **- Ex1:** There is an Ender-3 user saying that his temps are off by 20c when tested using a K type vs the stock when measured from within the hot end area. He bought two replacements and was having the same issue.

This is a incomplete statement. Without the temperature where the difference is 20°C, 20c is meaningless. When buing a random thermistor or setting a random table you get a random result. That's obvious.
Checking with a thermometer you 'trust' in is a good idea. Especially when combining random with random.
 
> [image]

A good answer! 
Greg tells us he is clueless.

> [image]

Greg shows us he is not that clueless - or?
If the temperature is 'constantly' off 20°C it really could be the reference resistor on the board what is off. However, a custom table will fix about every problem. However, using a 'electrically conductive' thermal compound is, in general, a not that great idea. It works with some **thermocoupl** systems but would be disastrous when shorting a reference **thermistor**.

> **- Ex2:** A video showing this issue: https://youtu.be/brl1weuD8bs

Excellent video, showing how to make a temperature table with a external reference thermometer. (Calibration of the reference thermometer could be improved by checking a second point - not only boiling water. In the table you can put the float values of the ADC-reading.)
Most difficult part of this is placing the the two sensors so, that they get the same temperature. That is much more reliable when both are not in/on the heater block but together in a stirred, heated bath of veritable oil.
When you have at least one thermistor, table, adc-pin combination you trust in, you can use that as a reference. Just connect you reference thermistor as usual and put the new thermistor on the other (hotend/bed) adc-pin. Now you'll get all the values side by side in the serial log. Then check a second time with the new table. Bed and nozzle temperatures should now be the same. Then test a third time with swapped thermistors and tables to exclude different reference (pullup) resistors.

> ...
> * Marlin documentation should have a well documented troubleshooting process for when this issue arises. How do we know when the thermistor or printer board thermistor input is operating correctly.

The web is full of this. The universal solution is trying to make your own table.
There is no better solution than measuring. Even when measuring the 
probability to mess that up is high. (f.e when measuring a shiny metal surface with a IR-thermometer) Without that you will always find only relative big errors.

> * Marlin 2.0 should be able to create its own thermistor table and save it to SD or output it as text.

As long as Marlin tries to run on the AVRs we have no memory for that rarely used features.
We already have a external [tool](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/buildroot/share/scripts/createTemperatureLookupMarlin.py) to create tables from 3 known points. In the web you can find other tools calculating tables, with similar formats, from datasheet values.

> * Marlin 2.0 should be able to tell you when a thermistor value may be wrong or show an error.

That's simply impossible without a reference.

InsanityAutomation, 2019-02-12:
The only extension to this area im looking to make is clamping MIN_TEMP and MAX_TEMP to the thermistor table limits to prevent dangerous scenarios when out of the table. Otherwise, I agree with the above. These machines also dont typically need the precision that comes with a honeywell data logger or something with multiple thermalcouples comparing data. Where you measure can also make a big difference.

To generate a table, I recently modified a spreadsheet macro somebody else did for Marlin 1.0 to work for the 2.0 format

https://www.thingiverse.com/thing:3300383

AnHardt, 2019-02-12:
> The only extension to this area im looking to make is clamping MIN_TEMP and MAX_TEMP to the thermistor table limits to prevent dangerous scenarios when out of the table.

That could be difficult for some of the 'handmade' tables having extra high/low values at the start and end of the tables. Best could be a limit temperatures from the datasheet - but i guess they are not available for all types.
What we can do, is to check and error, when we look up the ADC values for the min/max °C  temperatures - if we are 'outside' the tables.

InsanityAutomation, 2019-02-12:
@AnHardt My first thought was a check with a static assert in each table with preprocessors to just kill it there at compile time. The runtime check is a good thought too. Its about 4th on my list to look at.

I saw an MFG set min_temp below the table and max_temp above. When the temp went over the table (300c), it showed the first entry (-14c) and kept the heater on full blast.... Of course this had thermal runaway disabled as well. That was a gross misconfiguration and I was looking to prevent it. The reason for min_temp to be set low was removed with the change to only monitor when a heater is active, so now its unlikely they would have been open on both ends like that though.

AnHardt, 2019-02-12:
@InsanityAutomation 
Forget it.
We do initialize the min/max-ADC values from the opposite side. When the algorithm would't find a value, the MIN/MAX-TEMP_ERRORS would appear as soon as active.

InsanityAutomation, 2019-02-12:
@AnHardt Ill take a look if that behavior changed when I make it around here on my list. The scenario above occurred with a 2.0 snapshot from Sept.

InsanityAutomation, 2019-02-12:
@AnHardt also, to be clear I didnt physically see the failure, and I dont know if the thermistor was just melted to an open state in between. The datasheet rated it for 350c, which is where the max temp was set, but it was using a type1 table. User had reportedly set it to 315c. Ill reproduce at lower temps, but again not till I drill down to it in my list lol

Deneteus, 2019-02-12:
@AnHardt Thanks for the detailed response. You had the same exact reaction that I did.  I would think someone with his background would have figured it out himself since most of the people in the Ender 3 group are not tech saavy or electronically inclined.

I have used thermistors with water/oil cooling on the PC since the 2000, so I have experience with cooling and fans and like a few hundred thermal pastes as well. He was probably talking about something along the lines of Arctic Alumina or silver thermal pastes. Thermally conductive is what he should have been going for. 

There is alot of info about thermistors online.  I figured it would be more helpful not have to dig for it across multiple sites. 

The Ender-3 uses a 100K ohm NTC 3950FB that has a 1% tolerance.
5% tolerance means at 300°C an error of up to +/-15°C

I know that resistors can be off as much as their tolerance 10-20%. If his on-board series resistor was out of spec it would be off. The measurement can also be affected by electrical noise or a noisy unfiltered 
voltage source. (Which is why the capacitors had to be added to the Creality 1.1.2 boards.)

NTC Thermistors are only supposed to be off ± 2.0 °C. They have a resistance tolerance as well as a temperature tolerance. You can't directly read the thermistors resistance with an ADC though so you have to use a voltage divider.

I was actually thinking of building a rig to test thermistors using a lab calibrated heating source. That way I know for sure they aren't defective and my table would be more accurate since I would have its measured resistance.

I read that in Marlin Kimbra you can use the M305 command to adjust the the thermistor and ADC values. That would be a useful in this case if he just had the wrong thermistor.

I figured you were going to say you have no memory for that on AVR but on the newer 32-bit boards you do right?

@InsanityAutomation  I like the idea of using the thermistor table to keep the temp from going over what the thermistor can handle. That would kind be like what they already do with not allowing the nozzle to go below the bed after leveling.

**This is what started the thread:**
[image]

**This is what I said because it sounded more like an issue with the thermistor or his board. There has been no update since the conversation.** 
[image]

boelle, 2019-06-22:
missing label

zmp2000, 2021-06-19:
For the moment, what I did was: I created a Sheet with all thermistor tables values and created a coeficient values so all values get multiplied by the coeficient, I just copy and paste all values to my table