MarlinFirmware/Marlin #13132: How Marlin 2.0 works in STM32F103RC

Msq001, 2019-02-12:
I have changed the motherboard  and compiling environment to STM32F1, and compiled the firmware. bin file. I burned this file directly to the flash address 0x08000000 of stm32f103rc.  However,  Marlin 2.0 doesn't work. What should I do to make it work properly?

griffin-117, 2019-02-12:
Have you tried flashing to an SD and using that? As far as I know, Marlin 2.0 must be flashed through SD for 32bit boards.

AnHardt, 2019-02-12:
@griffin-117 
Nonsens.

Msq001, 2019-02-13:
thanks for your reply, I haven't the bootloader of stm32f103rc, and I didn‘t find the official bootloader of stm32f103rc, I programmed one by myself, and burned the firmware.bin to the address of 0x08005000 through  sd card, it also doesn't work.

Msq001, 2019-02-14:
I have solved this issue.  I judge whether Marlin2.0 works or not is that after reset, UART returns information or not . The pins I connect to uart is PA9 and PA10. so the SERIAL_PORT should be set to 1 in the configuration file. but i set to 0 before,  So I can't get feedback information on my computer. now I set 1 and I can get feedback now

closed 2019-02-14 by Msq001