MarlinFirmware/Marlin #13133: controller fan not working with software enable on TMC2130

Black6spdZ, 2019-02-12:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

LinFor, 2019-02-12:
What board and fan connections do you use? Looks like conflict in pin assignments

gloomyandy, 2019-02-12:
You will need to provide way more than that if you want to get help. Config files, pin files, what version of Marlin? What do you expect to happen, what actually happens.

Black6spdZ, 2019-02-12:
so is it supposed to work with software enable on SPI connected TMC2130s? on a Biqu SKR 1.1 connected to the fan port P2_03.. I set standard FAN_PIN to -1 and controller fan to correct fan pin assignment.. I'm under the impression as soon as any steppers start moving it should come on correct? I could just keep it as standard part cooling fan and set in start gcode to run.. just hate doing bandaids like this when marlin should properly control it.. extruder auto fan works a treat!

gloomyandy, 2019-02-12:
Posting your actual config files really does make it easier for people to understand what it is you are doing, because I'm still struggling to understand exactly what it is you have done. Isn't P2_03 used for the normal part cooling fan? Don't you need that? So config files and pins file please.

Also do you have any reason to believe that the software enable has anything to do with this problem? Did you have it working on the same hardware but using the normal enable mechanism?

Black6spdZ, 2019-02-12:
I just use a combination extruder/parts cooling fan
so in my pins file I set 
#define FAN_PIN            -1 //P2_03 
^ is this the correct to disable normal parts cooling fan operation?

in config_adv I set
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN P2_03 //-1        // Set a custom pin for the controller fan
  #define CONTROLLERFAN_SECS 0          // Duration in seconds for the fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 255        // 255 == full speed
#endif

I will try again this evening just to rule out a prior typo

Black6spdZ, 2019-02-12:
ok, so since I am not currently using my heated bed I wired the controller fan to P2_05.. now after resetting the controller it's running the fan even with steppers disabled.. so I'm pretty certain the routines used to sense enabled drivers isn't implemented for SPI enable yet

gloomyandy, 2019-02-13:
Try making the stepper enable values non-inverting.... Change
``` C
#define X_ENABLE_ON 0
#define Y_ENABLE_ON 0
#define Z_ENABLE_ON 0
#define E_ENABLE_ON 0 
```
To
``` C
#define X_ENABLE_ON 1
#define Y_ENABLE_ON 1
#define Z_ENABLE_ON 1
#define E_ENABLE_ON 1
```

Black6spdZ, 2019-02-13:
That did the trick! Is this a specific fix that applies to this particular board or any that use software enable with these particular drivers?

gloomyandy, 2019-02-13:
It applies to any board using TMC2130 drivers with software enable. It is not a fix just configuring things correctly for that mode of operation.

Black6spdZ, 2019-02-13:
Ah ok.. I would have never guessed that change would have worked for this particular problem. Did i miss something in comments or documentation?

gloomyandy, 2019-02-13:
Nope but I think all of the drivers that use software enable expect things that way. It basically comes down to what value they return for the enabled state, which in this case is 1.

gloomyandy, 2019-02-13:
I suppose it may get changed in the future so that people do not have to change the config, but it only really impacts code that needs to check the enabled state (and not much does that).

Black6spdZ, 2019-02-13:
yea, I was going to ask if future config might to checks on say tmc & software enable to flip that state.. so we can close this as its not a bug per se but a little idiot proofing would go a long way to help the project as a whole. Thanks guys!

closed 2019-02-13 by Black6spdZ