MarlinFirmware/Marlin #13134: Ender 3 with BLTouch and Bed size 235x235 issue

Skybeat, 2019-02-12:
Hello,

if i change the bed size for Ender 3 from 220 (in v.1.1.9) to 235 and have a BLTouch installed there is a issue.

ABL is not longer possible because withhin the bed leveling the X axis goes over the X_MAX_POS.

What i have done was:
set X and Y max pos to 
#define X_BED_SIZE 235
#define Y_BED_SIZE 235

set X and Y probe for the BLTouch 
#define X_PROBE_OFFSET_FROM_EXTRUDER -49  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -8  // Y offset: -front +behind [the nozzle]

How can i use the original Bed size of 235x235 with the BLTouch?

Thanks

Skybeat, 2019-02-12:
after a closer look, i have finde out the the RIGHT_PROBE_BED_POSITION value must be changed like this (if a BLTouch is used with a X_PROBE_OFFSET_FROM_EXTRUDER value:

If X_PROBE_OFFSET_FROM_EXTRUDER is a positiv value 
RIGHT_PROBE_BED_POSITION (X_BED_SIZE - MIN_PROBE_EDGE - X_PROBE_OFFSET_FROM_EXTRUDER )

If X_PROBE_OFFSET_FROM_EXTRUDER is a negativ value 
RIGHT_PROBE_BED_POSITION (X_BED_SIZE - MIN_PROBE_EDGE + X_PROBE_OFFSET_FROM_EXTRUDER )

boelle, 2019-03-06:
@Skybeat still working on this?

Skybeat, 2019-03-06:
> @Skybeat still working on this?

How do you mean it?
i have changed my copy of Marlin to test it, but i have still a issue.
In Cura (v.3.6.0) i use the original Ender 3 profile of cure. But if i try to place a peace to print at the front of the build blad in Cura,, the print himself prints not over the bed, it prints in front of the bed. Currently i don't know why.

boelle, 2019-03-13:
this does not seem to be an issue with marlin but with slicer software

try to attach your 2 config files so we can judge if its a marlin issue or not

remember to use bugfix 2.0 when you test

linvinus, 2019-04-14:
hi, i have the same problem. but i found where the problem come from.
in Conditionals_post.h we have
```
#define X_CENTER ((X_BED_SIZE) / 2)
#define Y_CENTER ((Y_BED_SIZE) / 2)
#define X_MIN_BED (X_CENTER - (X_BED_SIZE) / 2)
#define X_MAX_BED (X_CENTER + (X_BED_SIZE) / 2)
#define Y_MIN_BED (Y_CENTER - (Y_BED_SIZE) / 2)
#define Y_MAX_BED (Y_CENTER + (Y_BED_SIZE) / 2)
```
if bed size is odd number then because of rounding to integer 
Y_MAX_BED will be 1mm less then should.
(235) / 2 = 117
X_MAX_BED = 117+117= 234
as temporary solution change BED_SIZE to 234mm

i believe similar problem in #11635 and here https://reprap.org/forum/read.php?415,823273

also 
MIN_PROBE_EDGE must be defined as
```
#define MIN_PROBE_EDGE (max(abs(X_PROBE_OFFSET_FROM_EXTRUDER),abs(Y_PROBE_OFFSET_FROM_EXTRUDER))+10)
```

shitcreek, 2019-05-15:
@Skybeat Have you tried @linvinus 's solution?

closed 2019-07-11 by thinkyhead