MarlinFirmware/Marlin #13138: Creality CR10S-Pro - RepRapDiscount full grapic smart controller does not work

Brainc0re, 2019-02-12:
I'm trying to run the reprap full graphic display on my cr10s-pro.

A guy on thingiverse run a modified version of the reprap controller from a Anet A6 on his machine. But he does not respond anymore. So I hope someone can help me here.

The guy provided his version of marlin [https://drive.google.com/file/d/1LPwIqSGSMkrry8Z1kPjorwaBoMd7JQRw/view?usp=sharing](url)
and this pinout of the mainboard https://drive.google.com/file/d/1Ya5zHF_WQpw7XY_BoL0zs20UwexBvtwI/view?usp=sharing

I connected all pins with jumper wires because the header on the mainboard is reversed. For the smart controller, i used this pinout from the reprap site https://reprap.org/mediawiki/images/5/51/RRD_FULL_GRAPHIC_SMART_CONTROLER_SCHEMATIC.pdf

I checked his version and uncommented line 1764 //#define ANET_FULL_GRAPHICS_LCD and commented line 1669 #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

But the display wont work. Only the backlight is working. And when i turn the encoder, the backlight goes of for ~1sec and a blue led flashes on the mainboard...

I also need help with "LCDE and LCDRS" - i dont understand what that is and why there are two solderpads on mainboard for it.

I hope someone can help me. Thank you

InsanityAutomation, 2019-02-12:
Ive run a standard full graphics smart controller, just pulling the plastic off the LCD itself on the connectors to flip them, on the CRX and 10SPro without an issue.

1.1.9 Branch supporting the touchscreen if desired
https://github.com/InsanityAutomation/Marlin/tree/Creality_DWINTest

Oalf72, 2019-02-15:
catch bug on compilation: segmentation fault 
after analyze
error in line 46 in the file ultralcd_impl_DOGM.h
was: #include <U8glib.h>
need to: #include "U8glib.h"
after the fix everything compiled.
I took all the most recent and arduino, and libraries, and marlin.
set up under Prusa i3 Steel with FullGraphics
may be it for you too

boelle, 2019-03-06:
@Brainc0re still working on this?
tried bugfix 2.0?

Brainc0re, 2019-03-06:
@boelle

No. Its working now. I had to connect the solder points (R64 and R66).

https://imgur.com/a/igMochX

boelle, 2019-03-06:
@Brainc0re maybe give the close button below a small click :-D

closed 2019-03-06 by Brainc0re

Gongo1985, 2019-03-06:
Ciao puoi mettere qualche foto dei collegamenti r64 r66

Brainc0re, 2019-03-06:
In english please

boelle, 2019-03-06:
@Gongo1985 are you italian or spanish? just so i can pick correct language in google translate

boelle, 2019-03-06:
@Gongo1985 or you can translate: https://translate.google.com

boelle, 2019-03-06:
@Brainc0re but let us guess: he wants a photo of where/how you fixed it

boelle, 2019-03-06:
italian to english
[image]

Gongo1985, 2019-03-06:
Please can I take pictures how to connect the cables on resistors r64 r66 please please

boelle, 2019-03-06:
@Brainc0re i was correct, google translate is not the best so i think he wants a few pictures of what you did arround r64+66

Gongo1985, 2019-03-06:
Yessssss yesssssssss please pleaseeeeeeeeee

Gongo1985, 2019-03-06:
Yes yesss please please

boelle, 2019-03-06:
@Gongo1985 stai calmo :-D

Gongo1985, 2019-03-06:
Sorry🙏

Brainc0re, 2019-03-06:
No resistor is needed. Make a bridge with solder between the two points.

https://imgur.com/a/DQrWXld

Gongo1985, 2019-03-06:
I do the welding and then connect the 20-pin inverted cables

Gongo1985, 2019-03-06:
I do the welding and then connect the 10-pin inverted cables

Brainc0re, 2019-03-06:
Yes and no. EXP1 and EXP2 are inverted. And you need a Firmware where the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is enabled. But you can not take the firmware from my first post if you dont have a BLTouch installed.

Gongo1985, 2019-03-06:
Bltouch ok firmware ok Tomorrow I make the welds and I try again I hope to succeed you have had improvements with lcd

Gongo1985, 2019-03-06:
It works better With LCD

Gongo1985, 2019-03-07:
Hello guys I made the welds but it still does not work maybe I'm wrong the connection of the 10pin turns on the LCD but you can not see anything

Brainc0re, 2019-03-07:
Check the connection - pin by pin

https://imgur.com/6ZN5yvT

Its possible that you cable is horizontal AND vertical mirrored. For testing u can use EXP1 only

Gongo1985, 2019-03-07:
If I do 9 with 9 and 10 with 10 the screen stays off if I do 1 with 9 and 2 with 10 lights up

Gongo1985, 2019-03-08:
I finally managed many thanks to all of you thanks thanks thanks

Brainc0re, 2019-03-08:
What was the problem?

Gongo1985, 2019-03-08:
No problem everything works properly thanks to you

Gongo1985, 2019-03-09:
Hi guys and normal if I connect the printer with the USB cable the printer is not recognized