MarlinFirmware/Marlin #13139: With G29_RETRY_AND_RECOVER will not compile

5a2v0, 2019-02-12:
### Description

Error while compiling Marlin 2.0.x bugfix in Arduino IDE 1.9 beta (and also with Arduino IDE version 1.8.8) when using #define G29_RETRY_AND_RECOVER

### Steps to Reproduce

1. decomment **#define G29_RETRY_AND_RECOVER** in the Configuration_adv.h file
2. Try to compile

**Expected behavior:** compile without errors

**Actual behavior:** compiling will fail with errors

[config.zip](https://github.com/MarlinFirmware/Marlin/files/2855820/config.zip)

[image]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

InsanityAutomation, 2019-02-12:
I'm on mobile so I have not looked at your config files, but one thing to keep in mind is that this only supports bilinear and I believe mesh leveling. It does not support ubl.

5a2v0, 2019-02-12:
I'm using **#define AUTO_BED_LEVELING_BILINEAR**

InsanityAutomation, 2019-02-12:
Just took a closer look, can you try with this PR

https://github.com/MarlinFirmware/Marlin/pull/13039

Seems some of the changes to support it got in there in a way it wasnt ready for.

InsanityAutomation, 2019-02-20:
FYI, a couple fixes have been merged so please test this again and confirm.

Thanks

boelle, 2019-03-06:
@5a2v0 still working on this?

boelle, 2019-03-13:
@thinkyhead think we can close this one too :-)

closed 2019-05-16 by thinkyhead

npmomchev, 2019-09-20:
With G29_RETRY_AND_RECOVER option, Marlin will not compile again. I use last release of Marlin 2.0