MarlinFirmware/Marlin #13141: MINTEMP getting thrown on E1

nrgbistro, 2019-02-12:
### Description

When using a fresh install of this firmware on my TEVO Tarantula (completely stock), MINTEMP is thrown every time on E1 no matter what settings I change. The thermistor seems fine, as the temperature readout reads room temp and increases for a short time while the hotend heats up until MINTEMP is thrown. I am only using one hotend and one extruder.

### Steps to Reproduce

I am using Jim Brown's easy config. I have change almost nothing in the config (I have tried to change many different settings to no avail, but currently I just have an unmodified version of Jim Browns config installed). The configuration.h and configuration_adv.h files are attached below.

**Expected behavior:**

Printer heats up to selected temperature

**Actual behavior:**

Printer heats up for about 3-5 seconds, to no more than about 30 degrees celsius (according the the thermistor), then throws MINTEMP on E1

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files:
[configuration.h](https://drive.google.com/open?id=1gYZtRjoGxX8R2zBjFtMXGrqaJlOzBkwj)
[configuration_adv.h](https://drive.google.com/open?id=14jrTPPttyq0frj2pvxFmWgirZDIph0py)
* Provide pictures or links to videos that clearly demonstrate the issue:
[Printer Error](https://drive.google.com/open?id=1tI_1Y_4nAgeW_kTQVRIVyW6O2Dl0lPIe)

AnHardt, 2019-02-12:
Check the thermistor.
Check wiring - especially near the hotend. (because shortness of the time its likely close to the source of the heat)
MINTEMP usually means, thermistor disconnected.

nrgbistro, 2019-02-13:
@AnHardt how exactly do i check if it is functional? If it's reading room temp and the slightly heated temp somewhat accurately shouldn't that mean it works? There does appear to be some stripped wire coming right out of the hotend from the thermistor, but I couldn't tell if that meant it was broken or not. If it turns out to be broken, what type/size thermistor should I get as a replacement (for a TEVO tarantula)?

AnHardt, 2019-02-13:
> how exactly do i check if it is functional?

Move the **cold** hotend to a conveniently reachable place. Observe the display or M105 output in the log. The output should not 'jump' around a lot (more than +-1-2°C is bad but can be a noise problem). Touch the heater-block, not the wires, with your finger to heat it a bit. If the measurement is working the read temperature should rise. That may take a while. Now try to move the thermistor wires a bit, wiggle on them. If the read temperature changes a lot you found the problem. 
You will not get MINTEMP_ERRORs until you start heating. Start heating to about 1-2°C above the currently displayed temperature. Wiggle again.

> what type/size thermistor should I get as a replacement (for a TEVO tarantula)?

#define TEMP_SENSOR_0 ?
in **your** Configuration will give you the answer. A short description of the temperature sensors is given in the list right above that definition.
When buying i new one, you can replace ? with any **thermistor** from that list as long it will mechanically fit your heater block and has a table for 4K7 a pullup resistor.
For example 11 is highly available, small, cheap and well tested. (Don't forget a bit of non electrically conductive thermal compound when mounting it.)
----
This is not a Kindergarten. This is a bug-tracker and feature request tool for the big boys.
With beginners questions go elsewhere.

boelle, 2019-03-06:
@nrgbistro still working on this?

closed 2019-03-07 by thinkyhead