MarlinFirmware/Marlin #13142: Tmc2130 not working

Andreiva13, 2019-02-12:
Description
So I changed the board of my cr-10 with a mks gen l , changed firmware to marlin 1.1.9 , everything worked good on my bench so i got an old sever case which i modified to put the components in because i did not like the original case of the cr-10 after i did all the connections everything worked good until i realized i messed up the connection of the mosfet and the thermistor ( no damage there) so i switched them how they should have been and then the TMCs from x,y, and z stopped working , the extruder one still worked  
PS: I have redone the wireing , the firmaware worked , and they are powerd by a psu 
### Steps to Reproduce

**Expected behavior:** [What you expect to happen]
Everything to work as intented
**Actual behavior:** [What actually happens]
The tmcs don't work
[configuration .zip](https://github.com/MarlinFirmware/Marlin/files/2857127/configuration.zip)

boelle, 2019-03-06:
@Andreiva13 so you have not tried bugfix 2.0?

Andreiva13, 2019-03-06:
@boelle I checked that after I made the post

boelle, 2019-03-13:
did you try with bugfix 2.0?

Andreiva13, 2019-03-13:
Yes I did, I think I blown the drivers.

boelle, 2019-03-13:
yep, the are easy to blow

dont connect steppers or remove steppers with power on
or move steppers fast with the hand
or put them in the wrong way

maybe close this one and open new issue if the problem is still there when you get new drivers

closed 2019-03-13 by Andreiva13