MarlinFirmware/Marlin #13144: Byte-compiled GCode

MicroTransactionsMatterToo, 2019-02-12:
### Description

GCode, while effective, is an incredibly unwieldy format, that takes a lot of space and is inefficient. It is however easy to parse and industry standard for CNC type devices. However, I feel that a huge amount of space, and likely a decent amount of processing time could be saved by byte-compiling the gcode into a more efficient format. I'd be happy to work on a specification for it (I'm aware of a few others, but they're proprietary, and I'm not aware of any actual implementations). If I were to create this would it be likely it would be implemented into Marlin? 

### Original post I made on Reddit:
Unsure as to whether this already exists, but felt it might be worthwhile implementing if it doesn't. While GCode is simple, the files get pretty large pretty quickly for more complex prints. Given it's structure, it'd be pretty easy to create a binary version of it, with some structure like

&#x200B;

|Byte 1|Byte 2|Byte 3|Byte 4|Byte 5 - (Instruction Length - 4)|
|:-|:-|:-|:-|:-|
|Instruction Length|Instruction Category (M, G, T)|Instruction Number|Number of Arguments|Arguments|

The only issue I could see is instructions with multiple options for arguments, like the controlled arc move (G2-3) instructions.

I figured it might be advantageous to printers using less powerful hardware, perhaps removing the textual GCode parsing for bytecode parsing, which should be less expensive.

AnHardt, 2019-02-12:
https://reprap.org/wiki/Packed_Binary_GCode
https://github.com/repetier/Repetier-Firmware/blob/master/repetier%20communication%20protocol.txt

MicroTransactionsMatterToo, 2019-02-12:
Yup, taken a look at it. Are either of them well supported amongst different software? Because if the protocol/specification exists, but we're missing an example implementation/libraries, I'd be happy to remedy that.

AnHardt, 2019-02-12:
@bobc 
Is your [Packed_Binary_GCode](https://reprap.org/wiki/Packed_Binary_GCode) proposal in use somewhere?

@repetier 
Do you know if [repetier communication protocol](https://github.com/repetier/Repetier-Firmware/blob/master/repetier%20communication%20protocol.txt) in use outside the Repetier software family?

AnHardt, 2019-02-12:
Personally I prefer human readable communication. Without that debugging becomes a nightmare. Likely we have to support both of the protocols.
Currently communication and interpretation don't seem to be the limiting factor. Contributing yes, but not limiting Marlin. Planing short segments seems to be most demanding for the AVRs. 
 
Any suggestions about the 'back channel'? That's likely to be less critical. When really printing we just send 'ok's and occasional temperature reports, etc. Much less volume. But how about 'echo mode'?

MicroTransactionsMatterToo, 2019-02-12:
I wasn't thinking it'd be a primary form of communication, but an alternative for smaller devices to save space on the AVR controller, with the caveat that some things are not available when using it. I think the majority of people would just keep using normal GCode, but it'd be nice to have a specification and reference implementation for it so the option is there.

repetier, 2019-02-13:
I got some questions for implementation of repetier-protocol, so I think it is used somewhere else, most likely host variants. Don't know firmware not forked from repetier-firmware using it.

Some notes on binary input:
- It must be distinguishable from ascii gcode being send. In repetier-protocol first byte has bit 7 set to mark it as binary.
- Checksum should be better then xor, that is why we used fletchers sum.
- With our hosts it increases data throughput. That is because we can buffer more into the input buffer so usb-serial latency is not as much of a bottleneck then with ascii. Binary is around 50% of ascii with our format.
- In repetier-protocol the value format is predefined G, M, N 16 bit integer, T 8 bit, P,S 32 bit integer and the rest are floats plus one string allowed. This is for speed enhancement - we get binary in the correct number format and store it as such in a fixed class. Drawback is that the class uses memory for all letters.

Alternative might be to use a 6 byte struct to represend one letter and internal type that can return any type you want. If you then know usage is max. 10 letters you come away with 10*6=60 byte per command and can store them in a list. Or have a list for them and gcode points to first and count so you can even more reduce memory. Ascii can be parsed into this as well for faster access later. 

Binary format from Bobc looks also interesting as it is more flexible with types then repetier-protocol, also I'd change some points in format. But I like the type/format crunching and an optimizing host might even reduce 4 byte values to 1 byte is the value would fit. Firmware can then convert into target format.

MicroTransactionsMatterToo, 2019-02-14:
Well, I'll get started on some sort of implementation in C, although progress will be somewhat slow, as I have work and schoolwork to deal with as well.

gloomyandy, 2019-02-15:
I don't like to be negative but...

Is this really a good idea? Will it really actually save any program space? To do so you would have to build Marlin without any GCode support which means totally committing to this format. That means you can only communicate with Marlin using this format, so no simple terminal access and you need whatever host software you wish to use to support this new format fully. You will also need to rework all of the places in Marlin that use ascii strings to execute internal commands (like various menu options do). Then you will have to duplicate all of the validation code that currently parses the individual commands and unless you find a way to do this using the existing code will have to maintain a parallel set of code to the GCode parser for as long as both are in use. To do this just at the time that many people are beginning to switch to control boards that have 32bit cpus with native USB interfaces seems like it may be a lot of potentially wasted effort, effort that could perhaps be better spent supporting those 32 bit systems. It seems to me that systems like klipper already provide a good solution for using older cards.

I understand that it is your time and of course you are free use how you wish. But I do think that with features like this consideration has to be given to future maintainers. Having to support two gcode parsers/interpreters going forwards will add to the workload.

boelle, 2019-03-13:
@MicroTransactionsMatterToo is it relavant to keep this open?

MicroTransactionsMatterToo, 2019-03-17:
No, I'll close it now

closed 2019-03-17 by MicroTransactionsMatterToo