MarlinFirmware/Marlin #13145: Ramps EEB + RRD Fan Extender for extruder fan and model cooling fan + BLTouch

DangerD256, 2019-02-12:
I have ramps 1.4 with dual extruders and i want to connect 2 fans and BLTouch
1. Cooling fan for printing PLA controllable from LCD display and gcode.
2. Extruder fan auto on. (EXTRUDER_0_AUTO_FAN_PIN as i understood)
3. BLTouch
What should i change in configuration to make both work?
For printfan D4 is already set, for extruder i'll set EXTRUDER_0_AUTO_FAN_PIN 5
Also can i use BLTouch with that? As i read docs it also uses same port...

AnHardt, 2019-02-12:
Displace the fan extender two pins to the right. That will servo 0 and 1 let usable for servos/bl-touch

J-charles-C, 2019-02-14:
I,

http://www.3dmodularsystems.com/img/cms/RRD/RRDFanExtender_BLTouch.jpg

best

boelle, 2019-03-06:
@DangerD256 still working on this?

boelle, 2019-03-13:
@thinkyhead think we can close this one

DangerD256, 2019-03-13:
Changed pin in ramps pin file, all works fine except when I try to calibrate bed using bltouch it starts measuring outside of bed, but that's another story

boelle, 2019-03-13:
so if that is not related to this one maybe click close below?

closed 2019-03-13 by DangerD256