MarlinFirmware/Marlin #13147: MPCNC Dual Endstops, Y homing direction to posive does not stop when hitting endstops

wookie666, 2019-02-12:
### Description
[Marlin-MPCNC_Ramps_T8_16T_LCD_32step_DualEndstop.zip](https://github.com/MarlinFirmware/Marlin/files/2857876/Marlin-MPCNC_Ramps_T8_16T_LCD_32step_DualEndstop.zip)

Using dual endstops and dual motors, homing to Y+ does not stop when hitting Y endstops.
Pressing M119, i get Ymin, Y2min, endstops that trigger correctly when endstops are pressed manually
Changing Y homing direction to Y-, and reversing motor direction, homing stops correctly.

### Steps to Reproduce

Enable Y homing direction to 1 (y pos)
Enable dual motors and Dual endstops
G28 to home, Y moves to y positive

**Expected behavior:** [What you expect to happen]
When Y homes to positive, It should stop when hitting dual endstops

**Actual behavior:** [What actually happens]
When Y homes to positive, it doesn't stop when hitting endstops, and just continues stepping into endstops
#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

boelle, 2019-03-06:
@wookie666 still working on this?

wookie666, 2019-03-06:
Have up and changed to home y min. But would like to change this if possible

thinkyhead, 2019-05-16:
Dual Y endstops is not supported without also using independent dual Y motors.

closed 2019-05-16 by thinkyhead