MarlinFirmware/Marlin #13151: [bugfix 2.0] UBL on Anycubic Delta - clarification needed!

pandel, 2019-02-13:
Hi!

I am totally new to this whole UBL thing and as I am fighting my way through it I am facing some awkward problems I don't know how to solve. I think I might be doing things wrong, so I will describe what I have done so far and perhaps someone can tell me, what is going wrong here.

Printer: Anycubic Kossel Linear Plus with stock (switch) probe. Probe offset -15.90.

What I did right after flashing the firmware:

```
(attached the probe)
M502
M500
M501
G28
G33 P3
M500
G28
G29 P1 (3 touches per probe)
G29 P3 T ( have to execute it 2 times before every point is filled)
G29 S0
G29 A
(removed the probe)
```

At this point I noticed two things:
- two points in the mesh, front left and back left, seem to be interpolated much too high, the rest seems reasonable flat (from the values and from what OctoPrint BedVisualizer is showing)
- the complete mesh seems to be +0.4 -> I moved the nozzle to random points, slowly going to Z0 for the paper test -> at nearly every point I have to go to -0.35 to -0.4 until the nozzle grabs the paper

Even though I don't understand why this is the case, I set a Z offset for the whole mesh

`G29 P6 C-0.4`

and after that printed a mesh test with

`G26`

In my opinion the test mesh printed totally acceptable (btw, is it possible to enter some values for retract length and speed? - stringing is horrible). Some points seemed a _little_ bit squished and some a _little_ bit too high, but everything stuck nicely to the bed on the whole surface. And if I say a little, I really mean only a little!

Then I started to print a rather big round part (0.2 layer height, 110% first layer height) and suddenly the bed seemed to be tilted completely to the left! Left side won't stick at all, right side is squished all the way to the bed. I had to cancel the print.

Now I went through this whole procedure again a couple of times (and even trying to correct some points with `G29 P4 H0.1 R4`) but I couldn't get a reasonable print result. Small parts in the middle of the bed are ok, but the larger they get, it gets worse.

**What am I doing wrong??? Do I have to use `G29 J` additionally for some kind of tilt compensation? And how is `G29 P2` supposed to work (with probe, without probe, paper test, what???)**

BTW: I am a little bit afraid of issuing `G29 J` or `G29 P2` as I don't really understand what they will do and I don't want to crash my head into the bed or destroy my probe.

Regards,
Holger

pandel, 2019-02-13:
Ok, I went through the source code and here are my observations, please correct me, if I am wrong:

- `G29 J` needs a probe and probes either the three positions from the configuration or a rectangular grid and applies this as correction to the mesh
- `G29 P2` DOESN'T need a probe
- `G29 P5` might be a good addition to my "mesh is too high all over" problem to get a new starting point for the manual mesh correction
- `G29 P3` should possibly be changed to something like `G29 P3.10` to `G29 P3.13` for a different kind of weighting while extrapolation is done

Another strange effect:

When I execute a `G29 P4 H0.1 R4` and I _lower_ the nozzle too much, like from mesh point Z value -0.026 to -0.250 i. e., my stepper motors begin to rattle or vibrate or whatever, you can clearly hear the sound and feel the vibration when touching the belts. If you raise the value again, it goes away. That doesn't happen when I raise the nozzle. Could it be possible that one has to disable safe endstops so that this can work correctly?

Roxy-3D, 2019-03-05:
Please give an update....   And re-open if the problem is unresolved.

closed 2019-03-05 by Roxy-3D

pandel, 2019-03-05:
Here's my update: as I wasn't able to resolve my issues and because I bought a special, new Anycubic glassbed (Ultrabase), I first went back to bilinear, which gave more consistent (but not perfect) results. Then, after a little bit of bed clamp tuning, I don't need leveling at all. Just delta calibration, set Z probe offset and that's it.