MarlinFirmware/Marlin #13152: [bugfix 2.0] Wrong Z-Offset with UBL and Z Probe + Z endstop

fakcior, 2019-02-13:
Hello,
yesterday I was trying to work out Unified Bed Leveling, unfortunately without success.
My setup consists of mechanical Z-min limit switch + inductive Z probe connected to Z-Max endstop pin.

I'm trying to reproduce steps presented in UBL manual

M502            ; Reset settings to configuration defaults...
M500            ; ...and Save to EEPROM. Use this on a new install.
M501            ; Read back in the saved EEPROM. 
M190 S65        ; Not required, but having the printer at temperature helps accuracy
M104 S210       ; Not required, but having the printer at temperature helps accuracy
G28             ; Home XYZ.
G29 P1          ; Do automated probing of the bed.
G29 P2 B T      ; Manual probing of locations USUALLY NOT NEEDED!!!!
G29 P3 T        ; Repeat until all mesh points are filled in.
G29 T           ; View the Z compensation values.
G29 S1          ; Save UBL mesh points to EEPROM.
G29 F 10.0      ; Set Fade Height for correction at 10.0 mm.
G29 A           ; Activate the UBL System.
M500            ; Save current setup. WARNING: UBL will be active at power up, before any `G28`.
G26 C P5.0 F3.0 ; Produce mesh validation pattern with primed nozzle (5mm) and filament diameter 3mm 
G29 P4 T        ; Move nozzle to 'bad' areas and fine tune the values if needed
G29 S1          ; Save UBL mesh values to EEPROM.
M500            ; Resave UBL's state information.

After G28 homing with mechanical endstop my nozzle is just pieceof a paper above the bed, then after issuing G29 P1 I receive mesh with values about +1.5mm which results in printing in the air when doing G26 validation. 
I thought UBL will automatically calculate Z offset when using Z probe and Z min endstop, but this doesn't happen. 
How and when should I correct the Z offset?
Also does G26 prints with max feedrates? Is the way to slow it down?

landodragon141, 2019-02-17:
I recommend running G29 S1 and then M500 before running G26 just to make sure that it has been properly saved.

boelle, 2019-02-20:
@fakcior still having issues?

Roxy-3D, 2019-02-20:
> I thought UBL will automatically calculate Z offset when using Z probe and Z min endstop, but this doesn't happen.

With bed leveling turned off...  What happens when you do a G28?   Can you move the nozzle down so it just touches the bed?   What does the LCD display for a Z height?    If the nozzle is just touching the bed, the Z Height should read very close to 0.0000 mm.

> How and when should I correct the Z offset?

You should have the Z Offset set with a correct value before you try to bring up any of the bed leveling systems.    You should be able to do a small print in the center of the bed before you try to activate the bed leveling systems.    

> Also does G26 prints with max feedrates? Is the way to slow it down?    

It prints with a fraction of the max feed rates.   You can edit the code if it is going too fast.   Or reduce the specified max feed rates.

fakcior, 2019-02-22:
> With bed leveling turned off... What happens when you do a G28? Can you move the nozzle down so it just touches the bed? What does the LCD display for a Z height? If the nozzle is just touching the bed, the Z Height should read very close to 0.0000 mm.

It's homing with use of Z Endstop (not probe), so the nozzle is right above the bed and the Z height is 0mm. But when I activate UBL the Z height is set with use of Z probe. As a results it is about 2mm above the bed. 

I fixed the issue not by setting the Z offset but by issuing G29 P6 C 0.0 in order shift the Z height of the whole mesh and center in on 0mm.

boelle, 2019-02-22:
maybe attach 2 config files as zip file? that might tell @Roxy-3D a bit more

Roxy-3D, 2019-02-25:
> It's homing with use of Z Endstop (not probe), so the nozzle is right above the bed and the Z height is 0mm.

That adds a huge amount of complexity.   Is there a reason you are doing that?    The problem with doing this is you need the Z-Probe and the Z end stop to be exactly synchronized.   And it sounds like you have them mis-aligned by 2mm.

fakcior, 2019-02-25:
@Roxy-3D So how should I set the nozzle height and what for should I use Z endstop then? 
Thank you!

ManuelMcLure, 2019-02-25:
Generally, what users do when they have inductive probes is to hook the inductive probe up to the `Z_MIN` pin (as if it were the endstop) and set `Z_SAFE_HOMING` and `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`. This will ensure that the probe uses the middle of the bed as its homing location. Also make sure that `Z_MIN_PROBE_ENDSTOP` is _not_ set.

Roxy-3D, 2019-02-26:
> So how should I set the nozzle height and what for should I use Z endstop then?

If you have a Z-Probe...    Just ignore and don't use the Z-EndStop.    The way I set my Z_PROBE_OFFSET_FROM_EXTRUDER is I home the machine, and then very slowly, very carefully lower the nozzle until it just touches the bed.    And what ever that number deviates from 0.000mm, I subtract that number from Z_PROBE_OFFSET_FROM_EXTRUDER and reburn the firmware (and do a M502 & M500)

fakcior, 2019-02-27:
> If you have a Z-Probe... Just ignore and don't use the Z-EndStop. The way I set my Z_PROBE_OFFSET_FROM_EXTRUDER is I home the machine, and then very slowly, very carefully lower the nozzle until it [...]

Thanks. Maybe documentation should be updated with recommendation of not using Z endstop together with UBL?

pockata, 2019-02-27:
> After G28 homing with mechanical endstop my nozzle is just pieceof a paper above the bed

@fakcior I think this is your problem. After homing, your Z axis should be at 0 and the nozzle should be just a little bit above the bed, not at paper thickness (which can be 1-2mm thick). 

Before rewiring and re-flashing your firmware try the following:
1. Get a feeler gauge. It's cheap and a lot more accurate than a piece of paper.
2. Do a G28 (ensure there's no bed leveling active after that)
3. Lift Z to 0.3 (I usually use a 0.3 feeler gauge)
4. Move the nozzle to the middle of the bed and test with the feeler gauge. The nozzle should just barely scratch it.
5. Adjust the z offset if needed (I turn a screw on my machine)
6. Repeat steps 2-5 until you get a decent result 

With a properly set extruder offset and Z_PROBE_OFFSET_FROM_EXTRUDER, UBL and G26 test utility should work as expected. 

I run a similar setup to yours with a z-min endstop and a z probe and the bed leveling is working as expected.

Let me know how it goes.

boelle, 2019-03-06:
@fakcior still working on this?

fakcior, 2019-03-06:
> @fakcior still working on this?

As I stated earlier documentation should include some kind of notice for users with Z-probe and Z-endstop combo. They need to keep them in sychronisation or cancel the Z offset with G29 P5 Cx.x command (to zero the mean mesh height).

boelle, 2019-03-06:
never had an issue with probe and endstop

i even have dual z endstop and use a probe for bilinear probing, i did not even read the documentaion

Vertabreak, 2019-03-10:
i encountered many issues having a z end stop and a bltouch at the same time while using UBL.
issues stop when z end stop was removed allowing the bltouch to function properly.

boelle, 2019-03-13:
so maybe you can click close below?

closed 2019-05-17 by thinkyhead