MarlinFirmware/Marlin #13156: marlin 2.0  compilation problems

bogaskilop, 2019-02-13:
### Description

can not compile   marlin 2.0   some time ago i tested that marlin at early stage and i was able to compile it  for 8 bit mks gen l board  with arduino 1.9 now i downloaded new arduino beta  and new marlin 2.0 
and when i try to compile it  i get this : 

even if i will try to compile it with default configs i get same errors :( 

Arduino:1.9.0-beta (Windows 10), Płytka:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from c:\users\bogas~1.des\appdata\local\temp\arduino_build_131356\sketch\src\inc\marlinconfig.h:41:0,

                 from sketch\src\gcode\control\M3-M5.cpp:23:

c:\users\bogas~1.des\appdata\local\temp\arduino_build_131356\sketch\src\inc\SanityCheck.h:1972:1: error: non-constant condition for static assertion

 static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0

 ^

c:\users\bogas~1.des\appdata\local\temp\arduino_build_131356\sketch\src\inc\SanityCheck.h:1972:1: error: array subscript out of bound

exit status 1
In file included from c:\users\bogas~1.des\appdata\local\temp\arduino_build_131356\sketch\src\inc\marlinconfig.h:41:0,

                 from c:\users\bogas~1.des\appdata\local\temp\arduino_build_131356\sketch\src\gcode\gcode.h:266,

                 from sketch\src\gcode\host\M110.cpp:23:

c:\users\bogas~1.des\appdata\local\temp\arduino_build_131356\sketch\src\inc\SanityCheck.h:1972:1: error: non-constant condition for static assertion

 static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0

 ^

c:\users\bogas~1.des\appdata\local\temp\arduino_build_131356\sketch\src\inc\SanityCheck.h:1972:1: error: array subscript out of bound

In file included from c:\users\bogas~1.des\appdata\local\temp\arduino_build_131356\sketch\src\inc\marlinconfig.h:41:0,

                 from sketch\src\gcode\control\M280.cpp:23:

c:\users\bogas~1.des\appdata\local\temp\arduino_build_131356\sketch\src\inc\SanityCheck.h:1972:1: error: non-constant condition for static assertion

 static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0

 ^

c:\users\bogas~1.des\appdata\local\temp\arduino_build_131356\sketch\src\inc\SanityCheck.h:1972:1: error: array subscript out of bound

In file included from c:\users\bogas~1.des\appdata\local\temp\arduino_build_131356\sketch\src\inc\marlinconfig.h:41:0,

                 from sketch\src\gcode\temperature\M155.cpp:23:

c:\users\bogas~1.des\appdata\local\temp\arduino_build_131356\sketch\src\inc\SanityCheck.h:1972:1: error: non-constant condition for static assertion

 static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0

 ^

c:\users\bogas~1.des\appdata\local\temp\arduino_build_131356\sketch\src\inc\SanityCheck.h:1972:1: error: array subscript out of bound

Błąd kompilacji dla płytki Arduino/Genuino Mega or Mega 2560.

Ten raport powinien zawierać więcej informacji jeśli w 
File -> Preferencje zostanie włączona opcja "Pokaż
szczegółowe informacje podczas kompilacji"

Can any one point me what to double check ? 
 i didn't try platformio ide  yet

Ludy87, 2019-02-13:
Configfiles?

bogaskilop, 2019-02-13:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2860346/Marlin.zip)

dgrat, 2019-02-13:
My first guess, would be that the compiler used is incorrect. The code is correct.

RudyBenner, 2019-02-13:
Can't have 3 extruders on MKS_GEN_L

RudyBenner, 2019-02-13:
Once you have it compiling, you are free to start making changes to suit your installation.

bogaskilop, 2019-02-13:
Rudy  yes  i can i added  3rd extruder in config of mks gen l  and added external stepper driver pcb  mate .:P

bogaskilop, 2019-02-13:
btw i could not even compile  2.0 on default config  just unzipped  marlin 2.0 getting same errors :(

RudyBenner, 2019-02-13:
Cool. Thanks for that.
What pins?

bogaskilop, 2019-02-13:
what i found is  :
Error while detecting libraries included by C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_323305\sketch\src\lcd\menu\menu_advanced.cpp

Error while detecting libraries included by C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_323305\sketch\src\module\endstops.cpp

Error while detecting libraries included by C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_323305\sketch\src\module\probe.cpp

bogaskilop, 2019-02-13:
#define E2_STEP_PIN        42
#define E2_DIR_PIN         40
#define E2_ENABLE_PIN      63

just because i used  4, 5 and 6 pin for tmc2130 as CS spins

boelle, 2019-02-20:
@bogaskilop problem solved?

bogaskilop, 2019-02-23:
In file included from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\../inc/MarlinConfig.h:41:0,

                 from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\fwretract.cpp:27:

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\../inc/SanityCheck.h:1972:1: error: non-constant condition for static assertion

 static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0

 ^

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\../inc/SanityCheck.h:1972:1: error: array subscript out of bound

In file included from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\gcode\control\../../inc/MarlinConfig.h:41:0,

                 from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\gcode\control\../gcode.h:266,

                 from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\gcode\control\M120_M121.cpp:23:

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\gcode\control\../../inc/SanityCheck.h:1972:1: error: non-constant condition for static assertion

 static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0

 ^

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\gcode\control\../../inc/SanityCheck.h:1972:1: error: array subscript out of bound

exit status 1
In file included from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\dac\../../inc/MarlinConfig.h:41:0,

                 from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\dac\dac_dac084s085.cpp:7:

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\dac\../../inc/SanityCheck.h:1972:1: error: non-constant condition for static assertion

 static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0

 ^

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\dac\../../inc/SanityCheck.h:1972:1: error: array subscript out of bound

In file included from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:41:0,

                 from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\HAL\HAL_AVR\HAL_spi_AVR.cpp:40:

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1972:1: error: non-constant condition for static assertion

 static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0

 ^

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1972:1: error: array subscript out of bound

Error compiling for board Arduino/Genuino Mega or Mega 2560.

bogaskilop, 2019-02-23:
`In file included from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\../inc/MarlinConfig.h:41:0,

                 from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\fwretract.cpp:27:

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\../inc/SanityCheck.h:1972:1: error: non-constant condition for static assertion

 static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0

 ^

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\../inc/SanityCheck.h:1972:1: error: array subscript out of bound

In file included from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\gcode\control\../../inc/MarlinConfig.h:41:0,

                 from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\gcode\control\../gcode.h:266,

                 from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\gcode\control\M120_M121.cpp:23:

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\gcode\control\../../inc/SanityCheck.h:1972:1: error: non-constant condition for static assertion

 static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0

 ^

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\gcode\control\../../inc/SanityCheck.h:1972:1: error: array subscript out of bound

exit status 1
In file included from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\dac\../../inc/MarlinConfig.h:41:0,

                 from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\dac\dac_dac084s085.cpp:7:

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\dac\../../inc/SanityCheck.h:1972:1: error: non-constant condition for static assertion

 static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0

 ^

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\feature\dac\../../inc/SanityCheck.h:1972:1: error: array subscript out of bound

In file included from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:41:0,

                 from C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\HAL\HAL_AVR\HAL_spi_AVR.cpp:40:

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1972:1: error: non-constant condition for static assertion

 static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0

 ^

C:\Users\BOGAS~1.DES\AppData\Local\Temp\arduino_build_137161\sketch\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1972:1: error: array subscript out of bound

Error compiling for board Arduino/Genuino Mega or Mega 2560.`

bogaskilop, 2019-02-23:
no still getting this errors  and config is uploaded  i only added now  #define PARK_HEAD_ON_PAUSE  and still same errors :(

bogaskilop, 2019-02-23:
so what i observed if i only add 2nd or 3rd extruder then this problems occur   I just  compile it with only one extruder and now there wasn't any errors.

bogaskilop, 2019-02-23:
ok marlin 2.0 isnt good  XY tmc2130  is ok but Z on tmc2130 making strange noises

boelle, 2019-02-23:
24V ?

boelle, 2019-02-23:
checked vref?

bogaskilop, 2019-02-24:
@boelle  mate  its same config values moved from 1.1.9    to 2.0  especially abut  tmc drivers !!! 
besides  if you select more than 1 extruder it also not compile so  marlin 2.0 is not ready for test even comon !!

thinkyhead, 2019-02-25:
The sanity check was flawed. I have applied a patch. Please test the latest `bugfix-2.0.x` to confirm that the issue is resolved.

thinkyhead, 2019-02-25:
Here are some updated configs for you, @bogaskilop …

[bogaskilop-configs.tar.gz](https://github.com/MarlinFirmware/Marlin/files/2899209/bogaskilop-configs.tar.gz)

bogaskilop, 2019-02-25:
> Here are some updated configs for you, @bogaskilop …
> 
> [bogaskilop-configs.tar.gz](https://github.com/MarlinFirmware/Marlin/files/2899209/bogaskilop-configs.tar.gz)

thx you will check this when i get back home .

what a bout  retraction speed higher than 45 mm/s  on 1/32 stepper settings with lv8729 ? ( like in issue 12922 ) ??

thinkyhead, 2019-02-25:
> what a bout retraction speed higher than 45 mm/s on 1/32 stepper settings with lv8729 ? ( like in issue 12922 ) ??

Your stepping rates are limited by the MCU. Move too fast and the MCU gets no time to check the temperature sensors. If it can't check the sensors for 4 seconds it reboots the board for safety.

closed 2019-02-25 by thinkyhead