MarlinFirmware/Marlin #13161: [1.1.9] Using 2 individual Neopixel Strips (Caselight and "Printlight")

jb2304, 2019-02-13:
Hi everyone,
i'm trying to get 2 individual Neopixel-Strip working.
I got one strip on the Printer itself to light up the printing area. I also want to use it as "Printer-LED", that also shows the "printer events". 
It works as intended when i activate the lines in the codes.

The second strip is mounted in the case. I want to use this strip as case-light. It would just need to light up in the colour i set it to via the LCD-caselight-menu.
I tried "CASE_LIGHT_USE_NEOPIXEL" but with this option the caselight uses the same strip as the "Printer-LED". The second Strip does not light up at all.

Is it possible to use 2 Neopixel strips, controlled by different Pins on the board?

Thanks for any help.

7eggert, 2019-02-13:
There seems to be only one set of LED being supported, but in RGBW - if your case light was white only, you could make it work.

jb2304, 2019-02-13:
A while ago i had a setup with standard rgbw strips. It worked fine, but i required 8 PWM Pins to be fully controllable. But now i don't have 8 free pins in my current setup. That was the reason i wanted to switch to neopixels. I wanted to be able to control the 2 strips only using 2 pins.

7eggert, 2019-02-13:
I have a similar idea, so if you'd share your once-working config file, I might happen to actually implement it, but I'm thinking about using a PCA9685.

fred2088, 2019-02-16:
do you set to use different pin   ?
#define CASE_LIGHT_PIN 4 
#define NEOPIXEL_PIN    5

7eggert, 2019-02-17:
Currently I discovered the RGB PWM pins and your issue made me aware of possibly supporting more than one set of LED. I hoped to learn from your past experience.

jb2304, 2019-02-17:
Hi, yes i set the Case ans Neopixel Pin to different Pins. The commands where still using the same strip.

I've been through my files but i could not find my old build.
I basically definded 4 Pins for Standard Printer_LEDs (RGBW). For the caselights i defined another 4 pins in the Pins file and added a custom menu for it. So i was not using the standard caselight-menu.

fred2088, 2019-02-17:
Neopixel only need 1 pin to work with, case light and progress light need to be different pin...

boelle, 2019-02-20:
@jb2304 please use bugfix 2.0

1.1.9 has been put in the grave a long time ago

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-17 by thinkyhead