MarlinFirmware/Marlin #13162: Just trying to learn about adding new GCODE

n9jcv, 2019-02-13:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

n9jcv, 2019-02-15:
Hello, I am not sure this is the right place to post, 1st time posting, so please let me know if I made a mistake.

I made a fork of marlin v1.1.9.  I would like to use marlin for a robot of sorts.  I want to perform an action until a sensor triggers it to stop.  
I thought I could just make a new Mcode, then loop in the end code until my sensor says stop.  When I try to loop, about 3 seconds later marlin resets/reboots.  I believe it is the Watch dog timer killing it, although from serial I was not able to see this.  Please mind I am new to all of this.

Below is a sample of my code, maybe I did something wrong, or (most likely) there is a better way to do this;

`
inline void gcode_M444() {
//turns on my motor not a stepper
fanSpeeds[0] = 255;

int l = 1;
//switch closed is 0 - this would mean to stop
while (l > 0){
 if(READ(Z_MAX_PIN) == 0)
{lcd_setstatus("STOP");
//turn off motor
fanSpeeds[0] = 0;
(l==0);}
else
//switch open
{
lcd_setstatus("RUNNING!");
//}
//#endif
//endwhile
}
`
Any help you can provide would be great!  Thank You!

n9jcv, 2019-02-15:
One additional note, my new Mcode 444 would run this motor and no other G/Mcodes need to be executed while it is running.  When the sensor says stop, then I want the next Gcode to execute.

revilor, 2019-02-15:
First of all there is a bug in your code so it's actually an endless loop.

int l = 1;
//switch closed is 0 - this would mean to stop
while (l > 0){
if(READ(Z_MAX_PIN) == 0)
{lcd_setstatus("STOP");
//turn off motor
fanSpeeds[0] = 0;
**(l==0)**;}
else
//switch open
{
lcd_setstatus("RUNNING!");
//}
//#endif
//endwhile
}

Second, for any long-running operation you should call safe_delay from time to time to give the thermal manager a chance to do it's job. This will solve your issue with the watchdog, too.

So add safe_delay(10) to the end of your while loop and it should be ok.

n9jcv, 2019-02-15:
Thanks for the reply;  I am new to coding c++ as well
I don't see the endless loop, I see while l>0   and it is 1 before I start   so it executes while
then if the switch is closed, I set l = 0    so it goes back evaluates l and l is now zero so no longer executes while

Did I not understand something properly?

I am not sure safe delay solves my issue, looks like the above would be for 10 seconds.  I probably need to explain a bit more.
I want to run some gcode, steppers to move motors, then I get to a certain point, I just want steppers to stay where they are    then I execute my new m444   this would run a non stepper motor   for a period of time   1 minute, 10 minutes 60 minutes  or until the switch is pushed at say 13 minutes
when time is up or if the switch was pushed, then I would turn off non stepper motor and continue gcode    

i thought about using endstop/interrupts, but I dont want any gcode beyond my m444 to run   until time is up or switch is pushed

I hope my explination is clear
Thanks for your help

revilor, 2019-02-15:
safe_delay(10) delays for 10ms and also calls thermalManager which should avoid the watchdog to trigger. Out of security reasons (thermal runnaway protection) you should either call idle() or safe_delay()  on a regular basis for all long-running operations. 

The line marked in your code is no assignment but a comparision if l equals 0, so l will never get the value 0 and therefore the loop will never end. There is a difference between == and =.

revilor, 2019-02-15:
You can even get rid of the the l variable completely:
```
while (true) {
  if(READ(Z_MAX_PIN) == 0)
  {
    lcd_setstatus("STOP");
    //turn off motor
    fanSpeeds[0] = 0;
   // we are done here
    break;
  }
 else
 {
   //switch open
   lcd_setstatus("RUNNING!");
 }
 safe_delay(10);
}
```

n9jcv, 2019-02-15:
Thank you Revilor for the explanation, 

Is my idea feasible, to loop for 10, 30 or 60 minutes using safe_delay, and no further Gcodes would execute?

Is this idea so bad, that I should consider a different approach?
Like I say new to the coding and just starting to learn Marlin.  I do appreciate your assistance.

revilor, 2019-02-15:
In general I see no issue to do so.

Do you plan to send your gcode via USB? In that case you should also place the whole code block inside 

```
  KEEPALIVE_STATE(IN_HANDLER);
     ...
  KEEPALIVE_STATE(NOT_BUSY);
```

and enable HOST_KEEPALIVE_FEATURE so Marlin will send busy messages to avoid timeouts on the USB. 

And maybe you would like to use the LCD while your loop is running. In that case you should replace safe_delay by idle() so the LCD can update an respond to menu actions. You would end up with something like

```
inline void gcode_M444() {
  KEEPALIVE_STATE(IN_HANDLER);
  
  fanSpeeds[0] = 255; //turns on my motor not a stepper
  lcd_setstatus("RUNNING!");

  while (true) {
    if(READ(Z_MAX_PIN) == 0)
    {
     // sensor triggered, we are done here
      break;
    }
    idle();
  }

  lcd_setstatus("STOP");
  //turn off motor
  fanSpeeds[0] = 0;

  KEEPALIVE_STATE(NOT_BUSY);
}
```

n9jcv, 2019-02-15:
I will only use the sd card but i may want to use the lcd.  Thank you very much for the advice.  I hope after playing and learning marlin I may someday be able to contribute back and help improve but at this point I have much to learn.

Again much thanks you have been very kind
Bruce

closed 2019-02-20 by n9jcv