MarlinFirmware/Marlin #13165: [Question] Pause lcd menu

GMagician, 2019-02-13:
Not sure but now when you put printer in pause (M25) when you restart default selected item is "purge" and not "resume"...shouldn't be default "resume"?

InsanityAutomation, 2019-02-13:
There's a few reasons for the switch. Continue now has logic around the button to replace it with a disabled run out if the switch is still in the open state. Every Purge statement checks again and tries to reset the run-out condition. The change in order allowed us to use a single conditional block for both of those swapping buttons instead of identical conditions above and below. Also Purge Moore is the safe function where resume is not therefore it most likely should have always been at the top.

GMagician, 2019-02-13:
ok thanks for explanation

closed 2019-02-13 by GMagician