MarlinFirmware/Marlin #13167: Bed LEVELING LINEAR

ZS1FJK, 2019-02-13:
Hello everyone
First time posting and also not a pro but almost sure this is an issue. 

I have an Archim2 board i am building into a lulzbot taz 6. I have connected the hotend probe wire to the PB4_ECRSDV digital pin 102 also defined this in the configuration.h and the Bed corners are connected to GND. 

I start the print. After homing and wiping the nozzle the extruder will move to first probe point stop skip probing and then move to the second probe point were it will probe but does not stop. Then once it done trying to drill threw the bed corner it moves to the start position and starts warming the head for the print. It skipped the back two probe points. 

[Toaster Camera footage](https://www.youtube.com/watch?v=qoyLVM5lg0I)

[bed Leveling issue.zip](https://github.com/MarlinFirmware/Marlin/files/2862271/bed.Leveling.issue.zip)

Thank you for your attention

marcio-ao, 2019-02-14:
I don't know about the particular pin you have chosen, but we've found that not all pins work as they should because the proper mappings may not be in place. If you want a pin that works on the Archim2, I suggest pin #20, PB12 (Header J20 20) or pin #21 D21 PB13 (Header J20 19) since we are using that on our prototypes for servos with no problems.

The other alternative is to use the same pin for Z_MIN as Z_MIN_PROBE. Although our TAZ 6s for historical reasons have continued to use separate pins, its actually a easier to wire them together and not have to deal with two separate pins.

ZS1FJK, 2019-02-14:
Thank you for your reply marcio-ao.
I will give that a try tomorrow and give feed back if it solves the problem. Glad to hear you guys using the same board. Haha

ZS1FJK, 2019-02-15:
OK so i am now using pin #20 
These are my settings.

```
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN 
#define Z_MIN_PROBE_PIN 20            
#define Z_MIN_PROBE_ENDSTOP_INVERTING false  
#define Z_MIN_PROBE_ENDSTOP       
```

Now with these settings my Z dose not lift after homing and crashes into the bed.  Why did it stop lifting? I have defined.
`#define Z_AFTER_PROBING 8`

marcio-ao, 2019-02-15:
Use M119 to make sure that the Z_MIN and Z_MIN_PROBE are working properly. Push the button and make sure the output of M119 is correct. Use a wire to connect from a washer to the hotend and also make sure M119 shows the state changing properly.

marcio-ao, 2019-02-15:
You may also tweak ENDSTOP_NOISE_THRESHOLD. We use 2 in our FW.

ZS1FJK, 2019-02-15:
OK so 
`define# ENDSTOP_NOISE_THRESHOLD 2`
Z_MIN and Z_MIN_PROBE
Both are working

ZS1FJK, 2019-02-15:
OK so
```
#define ENDSTOP_NOISE_THRESHOLD 4
#define Z_MIN_PROBE_ENDSTOP_INVERTING true
```
Is working now
The Z is still not lifting after homing.

marcio-ao, 2019-02-15:
@ZS1FJK: Generally this means Marlin thinks the Z_MAX endstop is triggered. Like I said, M119 can help you troubleshoot this.

ZS1FJK, 2019-02-16:
I can confirm that Z_Max endstop is not triggered.

READ: x_min: open
READ: x_max: open
READ: y_min: open
READ: y_max: open
READ: z_min: TRIGGERED
READ: z_max: open
READ: z_probe: open

I forgot to mention that i changed my 
`#define Z_MIN_PROBE_ENDSTOP_INVERTING ` to true
 also
Im not sure if i should have these enabled

```
 #define ENDSTOPPULLUP_ZMIN 
 #define ENDSTOPPULLUP_ZMIN_PROBE 
 #define USE_YMAX_PLUG
 #define USE_ZMAX_PLUG

```
Bed leveling works if i let it drag itself threw the bed. I took of the wipe pad to stop it from crashing. Even if i just do a home on all the axis It dose not lift.

ZS1FJK, 2019-02-16:
I used the bed leveling debug to see whats happening. Its not being told to lift i think. 

READ:   destination=(-19.00, 258.00, 8.00) : Z_SAFE_HOMING
READ: >>> do_blocking_move_to(-19.00, 258.00, 8.00)
READ: <<< do_blocking_move_to
SENT: M105
READ: echo:busy: processing
busy: processing
READ: >>> homeaxis(Z)
READ: Home 1 Fast:
READ: >>> do_homing_move(Z, -405.00, [6.00])
SENT: M105
READ: echo:busy: processing
busy: processing
READ: <<< do_homing_move(Z)
READ: Move Away:
READ: >>> do_homing_move(Z, 2.00, [6.00])
READ: <<< do_homing_move(Z)
READ: Home 2 Slow:
READ: >>> do_homing_move(Z, -4.00, 1.50)
SENT: M105
READ: echo:busy: processing
busy: processing
READ: <<< do_homing_move(Z)
READ: >>> set_axis_is_at_home(Z)
READ: For Z axis:
READ:  home_offset = 0.00
READ:  position_shift = 0.00
READ: *** Z HOMED TO ENDSTOP (Z_MIN_PROBE_ENDSTOP) ***
READ: > home_offset[Z] = 0.00
READ:   current_position=(-19.00, 258.00, 0.00) : 
READ: <<< set_axis_is_at_home(Z)
READ:   current_position=(-19.00, 258.00, 0.00) : sync_plan_position
READ:   current_position=(-19.00, 258.00, 0.00) : > AFTER set_axis_is_at_home
READ: <<< homeaxis(Z)
READ: <<< Z_SAFE_HOMING
READ:   current_position=(-19.00, 258.00, 0.00) : sync_plan_position
READ:   current_position=(-19.00, 258.00, 0.00) : bracket_probe_move
READ: X:-19.00 Y:258.00 Z:0.00 E:0.00 Count X:-1910 Y:33189 Z:0
READ: <<< G28

```
#define Z_CLEARANCE_DEPLOY_PROBE 10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES 10 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE 10 // Z Clearance between multiple probes
#define Z_AFTER_PROBING 10 // Z position after probing is done //FJK

#define Z_PROBE_LOW_POINT -2 // Farthest distance below the trigger-point to go before stopping
```

Should `Z_AFTER_PROBING 10` not come into affect after homing?

Iv read allot of the other issues trying to understand the auto bed leveling and i think i have a very good grasp op how it works. Not bad for a noob HAHA. Anyway thank you for you help so far @marcio-ao really appreciate it.

ZS1FJK, 2019-02-16:
After some testing i discovered the following. In config 1 The probe will retract after all axis homing and config 2 it wont. I will list the two configs below. These are the only settings I change.

CONFIG 1
```
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
//#define Z_MIN_PROBE_PIN 20          
//#define Z_MIN_PROBE_ENDSTOP 

#define Z_CLEARANCE_DEPLOY_PROBE 4 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES 4 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE 4 // Z Clearance between multiple probes
#define Z_AFTER_PROBING 4 // Z position after probing is done //FJK
```
CONFIG 2
```
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_PIN 20          
#define Z_MIN_PROBE_ENDSTOP 

#define Z_CLEARANCE_DEPLOY_PROBE 4 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES 4 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE 4 // Z Clearance between multiple probes
#define Z_AFTER_PROBING 4 // Z position after probing is done //FJK
```
In simplify 3d I added a `G1 Z4 F3000` to my start script strait after the G29 and it works Z Lifts after homing. 
I think this mite be a bug. Could some one confirm i am correct or am i misunderstanding` #define Z_AFTER_PROBING 4 `.

[Configuration h.zip](https://github.com/MarlinFirmware/Marlin/files/2872212/Configuration.h.zip)

marcio-ao, 2019-02-18:
You want to check whether Z_MIN_PROBE_PIN is working. Lift the head up a bit so it does not touch the washers. Now, run M119 and verify it says "open". Now, run a jumper wire from the bed washer to the nozzle (wires with alligator clips are good for this test). Make sure it makes good contact on both the washer and the nozzle. Now run M119 and verify that is says "TRIGGERED".

Homing and probing are two separate things. G28 runs a home. It uses the Z_MIN_PIN. If homing works, and the nozzles pushes the button on the left side of the bed, then Z_MIN_PIN is good. G29 runs a probe. On the default TAZ 6, this uses Z_MIN_PROBE_PIN.

It is possible, with some rewiring work, to use Z_MIN_PIN for both homing and probing. This would be the case if you have electrically wired the homing button and the bed washers together, then you would enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and *only* define Z_MIN_PIN. I stress that this case requires some rewiring of the harness, however, as stock TAZ 6 keeps those on separate pins.

ZS1FJK, 2019-02-18:
Hello
G29 is working i had to put `#define ENDSTOP_NOISE_THRESHOLD 6` This solved my issues.

G28 is kind of working but the extruder does not lift after it finishes probing. So when it moves to do a wipe it hits the Z_MIN_ENDSTOP. 
https://www.youtube.com/watch?v=Ug28QXURuHk

boelle, 2019-02-20:
@ZS1FJK so issue is solved? maybe close this one

ZS1FJK, 2019-02-20:
I have found a work around by using a script but if you watch the vedio you will see the z dose not lift after homing. Should I close this? Or am I mistaken that it should lift after homing ?

boelle, 2019-02-20:
you should close since you found a way

ZS1FJK, 2019-02-20:
Thank you for your help @marcio-ao

closed 2019-02-20 by ZS1FJK

marcio-ao, 2019-02-20:
@ZS1FJK: I don't recall, but my guess is that it does not lift after homing. This needs to be done in the start GCODE, if that is what you want to do. It's possible we modify our firmware to lift after homing, for safety reasons. There are little tweaks like that we make to our FW to make the user experience a bit better.

BTW, I recently pushed a new version of Marlin 2.0.0 to our "devel" repo. One of the things I recently added was a build option for Unsupported_ArchimTAZ6, so if you wanted to use LulzBot FW for your Archim, that would be a good starting point. Be aware that it is untested, so you may still need to tweak it a bit. If you are using Z_MIN_PROBE, you would need to add "#define LULZBOT_WASHERS_ON_Z_MIN_PROBE" and configure "Z_MIN_PROBE", but otherwise the FW get you closer to the standard Lulzbot FW.

marcio-ao, 2019-02-20:
> It's possible we modify our firmware to lift after homing, for safety reasons.

Yes, I just looked at the code and I see we modify the FW to do this. We also modified the FW to move away from the endstops after homing when using sensorless homing. This can prevent chattering if homing is done twice in a row.

ZS1FJK, 2019-02-20:
@marcio-ao I would really like to see that marlin 2.0.0 release of yours. I looked on the lulzbot taz source but did not find it. Am i looking in the wrong place ?

marcio-ao, 2019-02-20:
Clone from here:

https://code.alephobjects.com/diffusion/MARLIN/

Then, "git checkout devel". Edit "Marlin/Configuration_LulzBot.h"

Change:

```
#define LULZBOT_Quiver_TAZPro
//#define TOOLHEAD_CecropiaSilk_AerostruderV2
#define TOOLHEAD_Quiver_DualExtruder
```

To:

```
#define LULZBOT_Unsupported_ArchimTAZ6
#define TOOLHEAD_Tilapia_SingleExtruder
```

Then, edit "Conditionals_Lulzbot.h". Change:

```
#if defined(LULZBOT_Unsupported_ArchimTAZ6)
    // Unsupported TAZ 6 with Archim 2. Washers must be
    // connected to Z_MIN (in parallel with homing button).
    #define LULZBOT_CUSTOM_MACHINE_NAME "Archim TAZ 6"
    #define LULZBOT_LCD_MACHINE_NAME "TAZ 6"
    #define LULZBOT_IS_TAZ
    #define LULZBOT_TAZ_BED
    #define LULZBOT_USE_ARCHIM2
    #define LULZBOT_USE_REPRAP_LCD_DISPLAY
    #define LULZBOT_USE_AUTOLEVELING
    #define LULZBOT_BACKLASH_COMPENSATION
    #define LULZBOT_USE_NORMALLY_CLOSED_ENDSTOPS
    #define LULZBOT_USE_MIN_ENDSTOPS
    #define LULZBOT_USE_MAX_ENDSTOPS
    #define LULZBOT_USE_HOME_BUTTON
    #define LULZBOT_ENDSTOPS_ALWAYS_ON_DEFAULT
    #define LULZBOT_USE_Z_SCREW
    #define LULZBOT_STEALTHCHOP_Z
    #define LULZBOT_HYBRID_THRESHOLD
    #define LULZBOT_BAUDRATE 250000
    #define LULZBOT_PRINTCOUNTER
    #define LULZBOT_MACHINE_UUID "a07987e3-7ca7-48e1-b7a4-cc2c45ff2742"
    #define LULZBOT_SDSUPPORT
    #define LULZBOT_USE_EXPERIMENTAL_FEATURES
#endif
```

To:

```
#if defined(LULZBOT_Unsupported_ArchimTAZ6)
    #define LULZBOT_CUSTOM_MACHINE_NAME "Archim TAZ 6"
    #define LULZBOT_LCD_MACHINE_NAME "TAZ 6"
    #define LULZBOT_IS_TAZ
    #define LULZBOT_TAZ_BED
    #define LULZBOT_USE_ARCHIM2
    #define LULZBOT_USE_REPRAP_LCD_DISPLAY
    #define LULZBOT_USE_AUTOLEVELING
    #define LULZBOT_BACKLASH_COMPENSATION
    #define LULZBOT_USE_NORMALLY_CLOSED_ENDSTOPS
    #define LULZBOT_USE_MIN_ENDSTOPS
    #define LULZBOT_USE_MAX_ENDSTOPS
    #define LULZBOT_USE_HOME_BUTTON
    #define LULZBOT_ENDSTOPS_ALWAYS_ON_DEFAULT
    #define LULZBOT_USE_Z_SCREW
    #define LULZBOT_STEALTHCHOP_Z
    #define LULZBOT_HYBRID_THRESHOLD
    #define LULZBOT_BAUDRATE 250000
    #define LULZBOT_PRINTCOUNTER
    #define LULZBOT_MACHINE_UUID "a07987e3-7ca7-48e1-b7a4-cc2c45ff2742"
    #define LULZBOT_SDSUPPORT
    #define LULZBOT_USE_EXPERIMENTAL_FEATURES
    #define LULZBOT_WASHERS_ON_Z_MIN_PROBE
    #define LULZBOT_Z_MIN_PROBE_PIN  ?? // Whatever pin here
#endif
```

We haven't tested this, so be cautious. If you find any problems, let me know and I might be able to add it to our FW for others to use.

ZS1FJK, 2019-02-20:
@marcio-ao OOOoo Exciting 
I am currently fixing up all my wiring and strapping everything in. Slso waiting on some heat sinks for drivers . Should be able to test everything by Monday if i get all my plugs...

marcio-ao, 2019-02-20:
@ZS1FJK: Our 2.0.0 branch is currently under heavy development, but our testing efforts are focused on unreleased products right now, so it is possible it won't work perfectly. That said, our 2.0.x is a lot closer to upstream Marlin that in it has ever been before, so you might have good luck with it.

If it works out well, let us know. It definitely would help us to have feedback as we move into qualifying 2.0.x for our older products. Good luck!

InsanityAutomation, 2019-02-20:
At least with the rambo, the latest builds seem to be doing fine on my Taz 6's fwiw

ZS1FJK, 2019-02-28:
@marcio-ao I have eventually finished the wiring. I ran out of connectors so had to order. Waited a week. Perks of living in a third world country :P. Any way i have manged to plug everything in and get the screen working. I never tested the SD card. Its not working I looked in the archim pins.h file and saw the SD Detect is only defined in the one screen.

`#if ENABLED(ULTRA_LCD)
  #define BEEPER_PIN       23   // D24 PA15_CTS1
  #define LCD_PINS_RS      17   // D17 PA12_RXD1
  #define LCD_PINS_ENABLE  24   // D23 PA14_RTS1
  #define LCD_PINS_D4      69   // D69 PA0_CANTX0
  #define LCD_PINS_D5      54   // D54 PA16_SCK1
  #define LCD_PINS_D6      68   // D68 PA1_CANRX0
  #define LCD_PINS_D7      34   // D34 PC2_PWML0

  #define SD_DETECT_PIN     2   // D2  PB25_TIOA0

  #if ENABLED(NEWPANEL)
    // Buttons on AUX-2
    #define BTN_EN1        60   // D60 PA3_TIOB1
    #define BTN_EN2        13   // D13 PB27_TIOB0
    #define BTN_ENC        16   // D16 PA13_TXD1 // the click
  #endif // NEWPANEL
#endif // ULTRA_LCD`

Should i open a new issue or am i misunderstanding this?

ZS1FJK, 2019-03-01:
> Clone from here:
>
> https://code.alephobjects.com/diffusion/MARLIN/
>
> Then, "git checkout devel". Edit "Marlin/Configuration_LulzBot.h"
>
> Change:
>
> ```
> #define LULZBOT_Quiver_TAZPro
> //#define [...]

I am not finding the 
LULZBOT_Unsupported_ArchimTAZ6 in the configuration_lulzbot.h file. has it been removed ?

marcio-ao, 2019-03-01:
It's not listed in Configuration_LulzBot.h, but you'll find it in Conditional_LulzBot.h

As for the pins file, make sure you are looking at pins_ARCHIM2.h rather than pins_ARCHIM1.h. Our FW has been configured for the Archim 2. If you want to use an Archim 1, you may need to change things a bit.

ZS1FJK, 2019-03-02:
> 
> 
> It's not listed in Configuration_LulzBot.h, but you'll find it in Conditional_LulzBot.h
> 
> As for the pins file, make sure you are looking at pins_ARCHIM2.h rather than pins_ARCHIM1.h. Our FW has been configured for the Archim 2. If you want to use an Archim 1, you may need to change things a bit.

@marcio-ao Sorry about that marcio-ao . I am new to the hole github and repository thing. I downloaded the GitHub desktop app but did not select the correct branch. HAHa . Ill drink more coffee from now on. :P

ZS1FJK, 2019-03-02:
@marcio-ao 
Marcio-ao I have found something but not sure on how to report it to you. Also i am compiling using Arduino IDE and its not compiling will see what i can figure out there. 
Thank you for you awesome support so far.

andrup502, 2019-05-25:
I'm getting a an "error compiling for board archim" message
I got arduino 1.9 beta and the archim addon but still no luck

ZS1FJK, 2019-05-25:
@andrup502

It's very difficult to help you without a config files or the error message. Have you read how to post questions ?

Hope you find a solution to your problem

andrup502, 2019-05-26:
Sorry, my error is 
In file included from sketch\src\Lulzbot_Extras.cpp:24:0:

sketch\src\module/stepper_indirection.h:59:26: fatal error: TMCStepper.h: No such file or directory

   #include <TMCStepper.h>

I assume this means I don't have the right library for the tmc stepper drive but every one i've tried doesn't, could you share what library you used to get it to compile?

ZS1FJK, 2019-05-26:
I would look at the wiki pages for the boards. On the archim 1 wiki there is a section with instructions for TMC driver install. https://reprap.org/wiki/Archim_v1.0