MarlinFirmware/Marlin #13169: [Q] Best practice to modify sanity checks for new kind of LCD

7eggert, 2019-02-14:
What is the best way to modify SanityCheck.h in order to compile for my custom display having
1) An LCD_I2C_TYPE_PCF8575 LCD2004 (not Zonestar / Anet)
2) A rotary encoder, 2 digital pins, button connected to ...
3) An ADC_KEYPAD

Intended target version is both 1.1.x and 2.0.x
(the former because of having a working config and thus starting there while making my LCD work)

My config for this display is:
```
+/**
+ * I²C-2004-Display and ADC key:
+ */
+
+// display
+  #define ULTIPANEL
+  #define LCD_I2C_TYPE_PCF8575
+  #define LCD_I2C_ADDRESS 0x27   // I2C Address of the port expander
+  #define LCD_WIDTH 20
+  #define LCD_HEIGHT 4
+  
+// encoder
+  #define NEWPANEL
+  #define BTN_EN1          11
+  #define BTN_EN2          10
+  #define BTN_ENC -1
+  #define REVERSE_MENU_DIRECTION
+  //#define REVERSE_ENCODER_DIRECTION
+
+// keys
+  #define REPRAPWORLD_KEYPAD
+  #define ADC_KEYPAD
+  #define ADC_KEY_NUM 8
+  #define ADC_KEYPAD_PLUS_ENCODER
```

My current hack to compile is:
```
--- a/Marlin/SanityCheck.h
+++ b/Marlin/SanityCheck.h
@@ -1467,7 +1467,7 @@ static_assert(X_MAX_LENGTH >= X_BED_SIZE && Y_MAX_LENGTH >= Y_BED_SIZE,
   + ENABLED(G3D_PANEL) \
   + (ENABLED(MINIPANEL) && DISABLED(MKS_MINI_12864)) \
   + ENABLED(MKS_MINI_12864) \
-  + (ENABLED(REPRAPWORLD_KEYPAD) && DISABLED(CARTESIO_UI) && DISABLED(ZONESTAR_LCD)) \
+  + 0 && (ENABLED(REPRAPWORLD_KEYPAD) && DISABLED(CARTESIO_UI) && DISABLED(ZONESTAR_LCD)) \
   + ENABLED(RIGIDBOT_PANEL) \
   + ENABLED(RA_CONTROL_PANEL) \
   + ENABLED(LCD_SAINSMART_I2C_1602) \
```

REPRAPWORLD_KEYPAD is needed to have the button #definitions

boelle, 2019-02-20:
Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

7eggert, 2019-02-21:
Maybe I should phrase it as "Bug: Configuring a custom LCD is prevented by sanity checks", but I felt that asking and fixing this myself would be better.
Discord is a white, empty page; facebook is blocked by DNS.

boelle, 2019-02-21:
> Discord is a white, empty page; facebook is blocked by DNS.

Where are you from?

7eggert, 2019-02-22:
Germany; using ScriptSafe; discordapp.com enabled but not sentry.io. Also using a DNS server blocking FB.

boelle, 2019-02-22:
are you blocking it yourself or are the internet provider blocking it? just sounds very strange

thinkyhead, 2019-02-23:
Hi @7eggert — Development guidance is one of the issues we don't mind being more helpful with, since we are the developers, and the other forums are meant mainly for end users who don't deal with the code.

To answer your original question, the sanity check is looking for the unique combination. If your top-level config option is named `LCD_I2C_TYPE_PCF8575` and this is another option that enables `REPRAPWORLD_KEYPAD`, then the new identifier for "`REPRAPWORLD_KEYPAD` by itself" becomes…
```diff
-  + (ENABLED(REPRAPWORLD_KEYPAD) && DISABLED(CARTESIO_UI) && DISABLED(ZONESTAR_LCD)) \
+  + (ENABLED(REPRAPWORLD_KEYPAD) && DISABLED(CARTESIO_UI) && DISABLED(ZONESTAR_LCD) \
+     && DISABLED(LCD_I2C_TYPE_PCF8575)) \
```
You'll also need to add:
```diff
+  + ENABLED(LCD_I2C_TYPE_PCF8575) \
```
So…
```cpp
#if 1 < 0 \
  + (     ENABLED(ULTIMAKERCONTROLLER) \
      && DISABLED(SAV_3DGLCD) \
      && DISABLED(miniVIKI) \
      && DISABLED(VIKI2) \
      && DISABLED(ELB_FULL_GRAPHIC_CONTROLLER) \
      && DISABLED(AZSMZ_12864) \
      && DISABLED(PANEL_ONE) \
      && DISABLED(MKS_12864OLED) \
      && DISABLED(MKS_12864OLED_SSD1306) ) \
  + (     ENABLED(REPRAP_DISCOUNT_SMART_CONTROLLER) \
      && DISABLED(REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER) \
      && DISABLED(LCD_FOR_MELZI) \
      && DISABLED(MAKEBOARD_MINI_2_LINE_DISPLAY_1602) \
      && DISABLED(MKS_12864OLED) \
      && DISABLED(MKS_12864OLED_SSD1306) ) \
  + (ENABLED(REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER) && DISABLED(BQ_LCD_SMART_CONTROLLER)) \
  + ENABLED(LCD_FOR_MELZI) \
  + ENABLED(MALYAN_LCD) \
  + ENABLED(MKS_12864OLED) \
  + ENABLED(MKS_12864OLED_SSD1306) \
  + ENABLED(MAKEBOARD_MINI_2_LINE_DISPLAY_1602) \
  + ENABLED(CARTESIO_UI) \
  + ENABLED(PANEL_ONE) \
  + ENABLED(MAKRPANEL) \
  + ENABLED(REPRAPWORLD_GRAPHICAL_LCD) \
  + ENABLED(VIKI2) \
  + ENABLED(miniVIKI) \
  + ENABLED(ELB_FULL_GRAPHIC_CONTROLLER) \
  + ENABLED(AZSMZ_12864) \
  + ENABLED(G3D_PANEL) \
  + (ENABLED(MINIPANEL) && DISABLED(MKS_MINI_12864)) \
  + ENABLED(MKS_MINI_12864) \
  + (ENABLED(REPRAPWORLD_KEYPAD) && DISABLED(CARTESIO_UI) && DISABLED(ZONESTAR_LCD) && DISABLED(LCD_I2C_TYPE_PCF8575)) \
  + ENABLED(LCD_I2C_TYPE_PCF8575) \
  + ENABLED(RIGIDBOT_PANEL) \
  + ENABLED(RA_CONTROL_PANEL) \
  + ENABLED(LCD_SAINSMART_I2C_1602) \
  + ENABLED(LCD_SAINSMART_I2C_2004) \
  + ENABLED(LCM1602) \
  + ENABLED(LCD_I2C_PANELOLU2) \
  + ENABLED(LCD_I2C_VIKI) \
  + (ENABLED(U8GLIB_SSD1306) && DISABLED(OLED_PANEL_TINYBOY2) && DISABLED(MKS_12864OLED_SSD1306)) \
  + ENABLED(SAV_3DLCD) \
  + ENABLED(BQ_LCD_SMART_CONTROLLER) \
  + ENABLED(SAV_3DGLCD) \
  + ENABLED(OLED_PANEL_TINYBOY2) \
  + ENABLED(ZONESTAR_LCD) \
  + ENABLED(ULTI_CONTROLLER)
  #error "Please select no more than one LCD controller option."
#endif
```

closed 2019-02-23 by thinkyhead

7eggert, 2019-02-23:
> are you blocking it yourself or are the internet provider blocking it? just sounds very strange

Blocking FB is sane.
Discord may or may be not blocked incorrectly by not allowing JS from sentry.io