MarlinFirmware/Marlin #13170: How to activate power loss recovery

mpm1396, 2019-02-14:
Hello guys,
I am not sure if am right here because I do not have an issue, just a question.
So I am running marlin 2.0 on my anycubic prusa i3. I´ve read this article here: http://marlinfw.org/docs/gcode/M413.html
But I am not sure what to do. Do I have to activate this: POWER_LOSS_RECOVERY in the firmware
or do I just have to activate this: M413 S1 with the command line?

mpm1396, 2019-02-14:
Edit: I want to use it so that it writes to the sd card. Without a power loss pin

AnHardt, 2019-02-14:
> POWER_LOSS_RECOVERY option must be enabled.

What's so difficult with that? If `POWER_LOSS_RECOVERY` is not activated in the config, it's not compiled into the binary, can't be de-activated by M413 because that's not in the firmware. Like with about any other feature of Marlin.

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration_adv.h#L726-L739

M413 S0 is most helpful in 'spiral vase mode'! Because Marlin does not now about layers it writes the recovery info on every change in Z. That would mean constantly.

boelle, 2019-02-20:
Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

thinkyhead, 2019-02-25:
> Marlin does not [know] about layers [so] it writes the recovery info on every change in Z.

By default, it saves the current state to the SD card every time the nozzle extrudes at a higher Z value than it has before.

closed 2019-02-25 by thinkyhead

niu3niu, 2019-07-25:
the resume() function cant rightly move z axis,
/*****************************************
dtostrf(info.current_position[Z_AXIS], 1, 3, str_1);
  //dtostrf(10.0, 1, 3, str_1);
  sprintf_P(cmd, PSTR("G1 Z%s F200"), str_1);
  gcode.process_subcommands_now(cmd);
********************/
these code cant rightly move  back to saved Z,i guess info.current_position[Z_AXIS] is wrong!!