MarlinFirmware/Marlin #13172: Help Wanted - Adding a second serial port, and defining it

tkward, 2019-02-14:
Hi,

Can anyone help with this please ?

I have the Biqu SKR1.1 and have compiled Marlin 2 from the manufacturers fork which compiles and works well.

[https://github.com/Msq001/Marlin-bugfix-2.0-biqu]

I have the MKS wifi module, and also the MKS TFT32. The TFT32 connects via AUX1 RX and TX, and the wifi module connects to the TFT32. This isolates the wifi RX and TX from the mainboard.

The MKS firmware on the wifi module is frankly rubbish, and it was suggested on another forum that I change the wifi module firmware to the EPS3D software. This then means that the wifi module won't work via the TFT32. I have done this, and it works really well when connected in place of the TFT32.

So, background over, here is what I need .....

I have been told that I can use both the TFT32 connected to AUX1, AND the wifi module if I connect it to one of the additional hardware serial ports on the MCU ( e.g P0.15 and P.16).

I have these defines in configuration.h

#define SERIAL_PORT -1
#define SERIAL_PORT_2 0

According to the datasheet : 

[image]

[image]

[My-config.zip](https://github.com/MarlinFirmware/Marlin/files/2865198/My-config.zip)

I have tried this but I can't work out how to enable those pins to be serial TX and RX. The lcd modules are Trinamic drivers NOT defined in my config, so P.0.15 and P.0.16 are not pre defined. Using either wifi OR the TFT32 works fine when connected to AUX1, but I need both connected at same time.

Does the number of the additional port ( 0 ) align with the physical port on the MCU ? 0 = UART 0,   
, 1 = UART1 ?

Do I have additional #defines, or changes to add to my pins_xxxx.h file ?

Thanks in advance,
Keith

fakcior, 2019-02-14:
You cannot do that since RX and TX on both AUX1 and TFT are connected to the same pins (P0.2, P0.3).
Source: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.1/blob/master/hardware/SKR-V1.1SchDoc.pdf

tkward, 2019-02-14:
I should be clear that I want to connect the wifi module to P0.15 and P.16 ... and leave the TFT32 connected to P0.2 and P0.3. How do I define this ?

gloomyandy, 2019-02-14:
Do you use USB access to your printer? If so what you are asking for is to have 3 serial ports. I don't think that Marlin supports that. If you do not use USB access to your printer you can try defining...
#define SERIAL_PORT 0
#define SERIAL_PORT_2 1
and this should tell Marlin to use two UART ports, UART port 1 should use pins P0_15 and P0_16. Note however that you may run into other problems as this is a rather unusual configuration and there is a chance that something somewhere may try and default SPI usage to those pins.

tkward, 2019-02-14:
Ideally, after configuring this I want to forego USB interaction unless for reflashing marlin. 

I want to do it this way to avoid having to use the mks firmware on their wifi chip. With their firmware the wifi module has to connect to tft32 module which then passes on commands to the MCU. Using the esp3d firmware gives a much better web interface but means having a direct connection to the mcu. Obviously without the additional serial port I would have to choose between the tft and the wifi module.

Thanks for the advice ... I will try it and advise.

p3p, 2019-02-14:
Making sure that the spi pins are disabled (not just undefined) in the pins file could avoid any conflicts caused be defaults being initialised.

```
  #define SCK_PIN           P_NC
  #define MISO_PIN          P_NC
  #define MOSI_PIN          P_NC
  #define SS_PIN            P_NC
```

gloomyandy, 2019-02-14:
Depends how things are setup for SD card access, if sharing the onboard SD card then there should be no need to set SCK_PIN etc. If you are not using Marlin's SD card access at all then the above defines are a good idea!

tkward, 2019-02-15:
Thanks for the help guys!!

Some interesting developments though.

I managed to get the second serial working as I wanted, and the wifi module can talk to the motherboard. I used Pins 0.0 an d 0.1 ( 2nd, and undefined, extruder step and direction pins to avoid SD card definition conflicts ). BUT, copying files to the SD card doesnt work. If I use same hardware ( and pin definitions ) with smoothieware it works happily. Also, I ended up using the reprap GLCD, and again with smoothieware ( with same pin definitions ) it works. With Marlin2 I get a lit but blank screen.

I know the answer is likely to be 'check pin definitions in configuration.h etc' but I, and 2 friends have gone over this 3 times already ). 

The only way around this ( due to time constraints ) is for me to use smoothieware, with the wifi on serial port 1, usb on -1, and using the GLCD. This gives me both extruders, wifi and a display. Yes, a different config than I orginally wanted, but it works.

gloomyandy, 2019-02-15:
You really need to supply the actual config and pins files you are using and details of what configuration it is you are actually trying to achieve. So for instance which SD card reader are you trying to use (the one onboard or the one attached to the display) and what display are you actually using (an actual link to the display you are using as there are many different ones that claim to be "reprap GLCD"). I have no idea why "copying files to the SD card doesn't work" but again you have not said how you are actually trying to do this? Is this by USB or from a PC based control program, nor have you said which SD card reader you are trying to use or what error you are getting. So if you still want to get this going with Marlin then more information is needed. If not and you are happy using smoothieware, please close the issue.

tkward, 2019-02-15:
I will add more details this evening.

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-15 by p3p