MarlinFirmware/Marlin #13175: Endstop doesn't trigger

Sixaxis9, 2019-02-15:
### Description
Hi everyone! I've just finished the assebly of my last 3d printer. It is completely custom built, including the electronics. All works fine, and I'm pretty happy with it, but I'm having some issues with the endstops. I set up my board file starting declaring:
`#define X_STOP_PIN        31
#define Y_STOP_PIN        30
#define Z_STOP_PIN        29`
And in configuration.h
`#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG

// Enable pullup for all endstops to prevent a floating state
//#define ENDSTOPPULLUPS //THERE ARE EXTERNAL PULLUPS ALREADY`

Since I only have min endstops and I've used external 10k pullup resistor.
I've checked with an oscilloscope the input pins, and they follow the switches correctly, but on Pronterface with the command M119 I see that the endstops never triggered. 
Casually, I tried to attach one endstop on pin 25, and it correctly triggers endstop Y. Is there any know issue about this? Tryed on Marlin 1.1.9, bugfix and also 2.0

**Expected behavior:**
When switch closes the system should notify that the endstop is triggered

**Actual behavior:** [What actually happens]
Running M119 command report it as always "open"

#### Additional Information
I reflashed the code with the PINS_DEBUGGING enabled and ran M43, here the output, which is correct:
`PIN:  29   Port: A5        Z_MIN_PIN                   protected
.                          Z_STOP_PIN                  protected
PIN:  30   Port: A6        Y_MIN_PIN                   protected
.                          Y_STOP_PIN                  protected
PIN:  31   Port: A7        X_MIN_PIN                   protected`

Sixaxis9, 2019-02-15:
Further information: I ran "M43 I" on pronterface and it reports correct readings for endstop pins. However, running M119 still gets wrong "open" or "TRIGGERED". Tried with 1.1.x bugfix.

landodragon141, 2019-02-15:
What is Y_STOP_PIN? Try changing it to Y_MIN_PIN in your board define.

landodragon141, 2019-02-15:
Looks like I'm ignorant and the STOP_PIN variant does seem to exist, but I still think you should try changing it to MIN_PIN. What type of board are you using?

revilor, 2019-02-15:
Wrong setting for *_MIN_ENDSTOP_INVERTING, maybe?

Sixaxis9, 2019-02-15:
Thanks for replying! I've tried also with Y_MIN_PIN, but it is the same (moreover, the "M43" command reports the right pin settings, so I don't think it is a pinout error). I'm using a custom board based on an Atmega 1284p.
Well, in my opinion even if they need to be inverted they should toggle the endstop status anyway. But when I send "M43 I" I receive the correct status of each pin, and if soon after I send "M119" the endstops status never change.

reloxx13, 2019-02-15:
* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
And for u the boards and pins files

Sixaxis9, 2019-02-15:
[files.zip](https://github.com/MarlinFirmware/Marlin/files/2869495/files.zip)
I've uploaded config files, board.h and pins.h where I added my board and also the actual board pin .h file.
Thanks for the support guys!

Sixaxis9, 2019-02-15:
Ok, I figured out this problem. Before opening this issue on GitHub I wrote on the [Marlin Forum](https://reprap.org/forum/read.php?415,847054,847115#msg-847115) and Dust, which I thank again, has solved the problem. 
There are two different pinouts for the 1284p chip, the "standard", and the "Sanguino" pinout, where the analog inputs are reversed. I used [this](https://github.com/MCUdude/MightyCore) set of libraries, and you can select which pinout you want to use in the "tools" section in the Arduino IDE.
Thank you all very much for the support given!

revilor, 2019-02-15:
You are welcome. Please close the issue now that it's resolved. Thanx.

closed 2019-02-16 by Sixaxis9