MarlinFirmware/Marlin #13176: LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI.

Santobert, 2019-02-15:
### Description

Enabling LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR. So you cannot set the progress shown in percent externally.

### Steps to Reproduce

1. Enable '#define REPRAP_DISCOUNT_SMART_CONTROLLER'
2. Enable 'LCD_SET_PROGRESS_MANUALLY'
3. Compile

**Expected behavior:** [What you expect to happen]
Compilation succeeds, M73 sets the current percentage shown on the display

**Actual behavior:** [What actually happens]
Compilation fails
```
In file included from sketch/src/feature/../inc/MarlinConfig.h:41:0,
                 from sketch/src/feature/tmc_util.cpp:23:
sketch/src/feature/../inc/SanityCheck.h:506:4: error: #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
   #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
    ^
In file included from sketch/src/feature/../inc/MarlinConfig.h:41:0,
                 from sketch/src/feature/caselight.cpp:23:
sketch/src/feature/../inc/SanityCheck.h:506:4: error: #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
   #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
    ^
In file included from sketch/src/core/../inc/MarlinConfig.h:41:0,
                 from sketch/src/core/../Marlin.h:24,
                 from sketch/src/core/utility.cpp:25:
sketch/src/core/../inc/SanityCheck.h:506:4: error: #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
   #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
    ^
In file included from sketch/src/gcode/../inc/MarlinConfig.h:41:0,
                 from sketch/src/gcode/queue.h:29,
                 from sketch/src/gcode/queue.cpp:27:
sketch/src/gcode/../inc/SanityCheck.h:506:4: error: #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
   #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
    ^
In file included from sketch/src/feature/../inc/MarlinConfig.h:41:0,
                 from sketch/src/feature/snmm.cpp:23:
sketch/src/feature/../inc/SanityCheck.h:506:4: error: #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
   #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
    ^
In file included from sketch/src/core/../inc/MarlinConfig.h:41:0,
                 from sketch/src/core/language.h:24,
                 from sketch/src/core/serial.cpp:24:
sketch/src/core/../inc/SanityCheck.h:506:4: error: #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
   #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
    ^
In file included from sketch/src/feature/../inc/MarlinConfig.h:41:0,
                 from sketch/src/feature/I2CPositionEncoder.cpp:31:
sketch/src/feature/../inc/SanityCheck.h:506:4: error: #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
   #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
    ^
exit status 1
In file included from sketch/src/feature/../inc/MarlinConfig.h:41:0,
                 from sketch/src/feature/baricuda.cpp:23:
sketch/src/feature/../inc/SanityCheck.h:506:4: error: #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
   #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
    ^
In file included from sketch/src/feature/../inc/MarlinConfig.h:41:0,
                 from sketch/src/feature/closedloop.cpp:22:
sketch/src/feature/../inc/SanityCheck.h:506:4: error: #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
   #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI."
    ^
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.
```

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

Ludy87, 2019-02-15:
Enable LCD_PROGRESS_BAR

Santobert, 2019-02-15:
That worked. Thank you.

closed 2019-02-15 by Santobert