MarlinFirmware/Marlin #13177: adding extra feedback to movement completed

amrithmmh, 2019-02-15:
if i send G0 100 there is no feedback saying it is completed from marlin is there a way to add it to marlin 
i am modifying marlin to use for my college project 
i need an OK or done  after each moves like X Y or XY combined so that i can use servo M280 command and then angain movement etc

amrithmmh, 2019-02-15:
can anyone kindly point me in the right direction just to add an extra serial print statement after doing stepper motor movement absolute movement G0 X100 Y50 i seached marlin code but its very huge and i am unsure where to add which function does the actual stepper motor movement?

ManuelMcLure, 2019-02-15:
A `G0` will queue the command and return immediately, whether the movement has completed or not. What you need is to add a `M400` gcode any time you need to synchronize with the actual movement. The `M400` gcode will wait until all queued movements are complete before returning OK.

amrithmmh, 2019-02-18:
@ManuelMcLure so after G0 i can add M00 so can get a feedback as soon as the movement is completed?
also i think i found a similar issue can you merge it with the same isuue
[https://github.com/MarlinFirmware/Marlin/issues/5783](issue 5783)

ManuelMcLure, 2019-02-18:
Correct. What you can do in your gcode is something like this:
```
G0 X210 Y20
M400
M280
```
That way you're assured that the `M280` will not be executed until the `G0` command has completed and the head is actually at  210x20.

boelle, 2019-02-20:
Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

amrithmmh, 2019-02-20:
@boelle  thanks i didnt know a discord group exists
 i will cloase this issue
thanks @ManuelMcLure

closed 2019-02-20 by amrithmmh