MarlinFirmware/Marlin #13178: Pause LCD Menu - purge more causes continue

bjbeare, 2019-02-15:
### Description

Filament Change menu item is confused.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

bjbeare, 2019-02-15:
I don't see a problem with the code... and this looks like a 'stale build data' kind of problem. I cleaned out my temp directories in case something is being cached... but could well have missed something. I use the Arduino IDE.

InsanityAutomation, 2019-02-17:
Can you confirm which date your snapshot was from? Ill test it again on this end with the current snapshot, But I want to be sure that you were working with a build after host prompt support was merged.

bjbeare, 2019-02-17:
commit e00aad84afe1660bc27da6e3ea4db1f73a183474 (origin/bugfix-2.0.x)

InsanityAutomation, 2019-02-17:
Thanks, can you post your configuration files as well? Just bumped my build to today and ill be flashing again shortly.

InsanityAutomation, 2019-02-17:
My machine showed the same symptom. Found the problem and ill have a PR opened to fix it shortly.

boelle, 2019-02-20:
@bjbeare so your issue is solved? maybe close this one

bjbeare, 2019-02-25:
@boelle 
Verified. Thank you for the fix.

closed 2019-02-25 by bjbeare