MarlinFirmware/Marlin #13179: Firmware V2.0 not saving settings

TCWC, 2019-02-15:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

TCWC, 2019-02-15:
[X3A Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/2870447/X3A.Config.Files.zip)

GMagician, 2019-02-15:
M500 must be executed if you change settings and want them to be saved to eeprom

TCWC, 2019-02-15:
> M500 must be executed if you change settings and want them to be saved to eeprom

Thanks for responding. I Did do the M500 From repetier Host but nothing saves.

ManuelMcLure, 2019-02-15:
Also, 

> My XYZ alwasys goes back to 0 and flashes

is normal. The printer has no idea of what has happened to the head/bed since the power was cut off - they could have been moved manually. It requires a `G28` to home so that it knows where to reference XYZ.

TCWC, 2019-02-15:
> Also,
> 
> My XYZ alwasys goes back to 0 and flashes
> 
> is normal. The printer has no idea of what has happened to the head/bed since the power was cut off - they could have been moved manually. It requires a G28 to home so that it knows where to reference XYZ.

The Factory Firmware didn't do this. Maybe I'm suppose to add some kind of script the runs when it boots up?? Not sure how to do that yet either. But also If I do a auto home the values pop up but if it idles to long they will start to flash as well. Which never did this before either.

ManuelMcLure, 2019-02-15:
If the factory firmware did not require a home before having "valid" XYZ values it was a bug. There is literally no way for the printer to know what the XYZ position is after a reboot without homing, unless it's a very expensive closed-loop servo system instead of the open-loop system used in 99% of inexpensive 3D printers.

The loss of position when idle is probably due to the `DEFAULT_STEPPER_DEACTIVE_TIME` setting (default 120 seconds). This makes Marlin deactivate the stepper 120 seconds after the last move. Since the axis can now be moved by hand, the position of the axis is marked as invalid since it could change without Marlin being aware of it. You can set `DEFAULT_STEPPER_DEACTIVE_TIME` to 0 to disable this.

ManuelMcLure, 2019-02-15:
> Maybe I'm suppose to add some kind of script the runs when it boots up??

The gcode generated by your slicer will probably include a `G28` call to home the printer before starting a print. Generally the position of the head is unimportant until you actually start printing something, so this is usually sufficient.

TCWC, 2019-02-15:
I Forgot about the stepper deactivate time.. I will correct that. I know your right about having a vaild XYZ. But now if I remember correctly when the printer was turned on from power supply it would do a auto home no matter where it was. Can I put G28 somewhere in the start up of the printer??

GMagician, 2019-02-15:
@ManuelMcLure 

> unless it's a very expensive closed-loop servo system instead of the open-loop system

this will not save you from homing. Closed loop let you know where you are after an home has been done. It you want a "homeless" system you an absolute encoder (and this will be more expensive than a simple closed loop)

ManuelMcLure, 2019-02-15:
> Can I put G28 somewhere in the start up of the printer??

I don't think there's a configuration option to do this. But I have to ask "why do you want this?" I can count on the fingers of one hand the number of times I've had to manually home my printer in the last six months or so - I just let the startup code in my gcode files do the homing for me when I start a print.

ManuelMcLure, 2019-02-15:
>It you want a "homeless" system you an absolute encoder

Yup. My mistake. In any case, something that's not going to be on a $200 price-class printer.

TCWC, 2019-02-15:
Ill try it out. Thanks

revilor, 2019-02-15:
I don't think it's a good idea to auto-home on power on. But it's your printer that might get hurt in the end, so if you really want to do so and your printer has an SD card slot you can try [autostart](https://github.com/MarlinFirmware/Marlin/wiki/Autostart)

TCWC, 2019-02-15:
> I don't think it's a good idea to auto-home on power on. But it's your printer that might get hurt in the end, so if you really want to do so and your printer has an SD card slot you can try autostart

That's kind of the issue I'm having. My SD Card doesn't initialize on boot up. I have to go into the printer display and initialize the card. Old Marlin firmware did it automatically.

ManuelMcLure, 2019-02-16:
Take a look at [https://github.com/MarlinFirmware/Marlin/issues/11788](https://github.com/MarlinFirmware/Marlin/issues/11788) - there are a couple of changes that might help with SD card initialization.

closed 2019-03-01 by TCWC