MarlinFirmware/Marlin #13182: [Bugfix 2.0] STM32F4 HAL compilation on a Nucleo F401RE as a STEVAL Board

youprint, 2019-02-15:
I'm trying to compile Marlin 2.0 on a Nucleo F401RE (basically the same as 401VE from STEVAL printer board) with 2 X-Nucleo IHM02A1 (L6470) stacked and one X-Nucleo IPS02A1 (VPS253SH) to drive the extruder and the hotbed.
I tried to compile on Arduino 1.8.8, 1.9 Beta, Atom and VS Code but i can't compile it. 
I used the last STM32duino core, the last Marlin 2.0 and last 6470 drivers.
i tried a repo that i found on Github but no success : 
https://github.com/Bob-the-Kuhn/Marlin-Bob-2/tree/L6470-bugfix
 
please help me

Bob-the-Kuhn, 2019-02-16:
Take a look at issue #12824.  At that time no one officially supported the F401.  In there is a pointer to an Arduino library that will compile.

Never got feedback on if the compiled image actually ran.

youprint, 2019-02-16:
Thank you.
I was thiking that STEVAL was supported, that's too early for now. i have a lot of STM32 dev board, so i could help testing. I tried the STM32F1 hal (ZET6) with success on a dev board using MKS robin config. GRBL is now working with powerstep01 and L6470 on a nucleo F411 and hope it will be supported one day in Marlin. 
Thanks for your work Bob, if you need beta testers i'm here ;-)

Bob-the-Kuhn, 2019-02-18:
What isn't supported in Marlin?

STEVAL and L6470 are supported.  Powerstep01 isn't supported because I have not taken the time to properly re-write the library.

boelle, 2019-02-20:
Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

thinkyhead, 2019-02-25:
We can't support compile errors in forks of Marlin. The code you reference does not exist in this fork.

closed 2019-02-25 by thinkyhead