MarlinFirmware/Marlin #13184: Disabling PS_DEFAULT_OFF not working as expected

kAdonis, 2019-02-16:
### Description

I am using an ATX PSU for a long Time now powering my RAMPS/Re-Arm with 5V Standby
I have:
`#define POWER_SUPPLY 1`
and I had always enabled PS_DEFAULT_OFF and AUTO_POWER_CONTROL
But now I'm using TMC5160 drivers which need to have 12V (or 24) when Marlin starts, because they dont get configured when not powered with Vm (12V)( I will open a separate Issue for this)
I tried to disable  PS_DEFAULT_OFF
` //#define PS_DEFAULT_OFF`
But the PSU stays off after a reset or powercycle, the behavior is the same as with  PS_DEFAULT_OFF enabled, with one exception: it does **not** react to M80 until a M81 is sent

### Steps to Reproduce

  1. Use a switchable PSU disable PS_DEFAULT_OFF and enable AUTO_POWER_CONTROL
  2. powercycle your printer

**Expected behavior:**
  The PSU is switched on for the configured time (POWER_TIMEOUT)

**Actual behavior:**
  The Psu stays off

using bugfix 2.0.x (e00aad8) 15. feb. 

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/2872052/Configs.zip)

kAdonis, 2019-02-16:
I found a Hack:
in Marlin.cpp from line 218

```
void setup_powerhold() {
  #if HAS_SUICIDE
    OUT_WRITE(SUICIDE_PIN, HIGH);
  #endif
  #if HAS_POWER_SWITCH
    #if ENABLED(PS_DEFAULT_OFF)
      PSU_OFF();
    #else
      PSU_ON();
    #endif
  #endif
}
```
I changed `PSU_ON();`
to `PSU_PIN_ON();`
this switches the PSU on at start for about 2 seconds, which is enough to get the TMC5160 configured

GMagician, 2019-02-16:
Check if you have PS_ON_PIN defined somewhere

thinkyhead, 2019-02-20:
> it does **not** react to M80 until a M81 is sent

In that case, Marlin must be setting its initial PSU "belief" incorrectly. Your "hack" is not a bad solution. I'll just check to see if there's a simpler one.

thinkyhead, 2019-02-20:
Ok, I've patched the issue. It should now be fixed for all.

closed 2019-02-20 by thinkyhead

kAdonis, 2019-02-20:
Great! Thank you!

commit 89bbc30a (2019-02-20): Fix init of power supply pin

Fix #13184