MarlinFirmware/Marlin #13186: Marlin 2.0 xyz calibration cube surface finish against 1.9.1

engelbja, 2019-02-16:
Good day all
Have posted this at https://reprap.org/forum with no replys.

Have downloaded bugfix-2.0.x from [marlinfw.org] 2weeks ago
When printing a xyz calibration cube I get surface problems as per attached picture.

This defect will be printed when the xyz cube are exactly in the middle of my delta printer build plate and only on the Y axis - when I move it 30mm left/right/front/back - no problems. This told me that there are a mechanical problem. The anycubic linear plus are 2 months old, but I removed my rails, cleaned & oiled them (smooth as butter) and also checked, motor pinions and belts. Exactly the same problem happening after this.
I moved back to marlin 1.1.9 and no issues - surface finish are 100%

My feed rates/acceleration/jerk are copied from 1.1.9 and I basically copied all my other settings from 1.1.9 to 2.0.
I use exactly the same gcode.
1st picture are 1.1.9 and 2nd are 2.0 ( The xyz cubes were only printed 50% because the problem are at the bottom - On the side you will see where the Y start, just beneath it are the problem)
[image]
[image]

landodragon141, 2019-02-17:
Please upload a copy of your configuration files.

engelbja, 2019-02-17:
Sorry here it is

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2872970/Marlin.zip)

closed 2019-02-25 by thinkyhead