MarlinFirmware/Marlin #13188: RE-ARM, RAMPS 1.6plus, and PT100

khomouda, 2019-02-17:
I recently upgraded my CoreXY printer with Re-ARM and RAMPS 1.6 PLUS and installed TMC2130 drivers in it.
the main changes were:
*24v power supply with no extra changes except a voltage converter for the fans.
*BLTouch does not work off the servo 3.3v so I had to connect its power supply to 5v in J3 in the RE_ARM
The PT100 works off A9 in AUX2 in the Ramps, which supply 3.3v so as not to burn the MCU, but I had to call it 3 in the pins file for it to be recognized, but,  at room temperature, the bed is reading 22 and the PT100 is 24, when heating it goes quickly to 270 and halts the printer, although my DMM temp probe reads 95 at 200 of PTT, I think the temp table needs changes with voltage changes﻿ (0.8v at 24degree@3.3v vs. 1.2v at 24degree@5v)

landodragon141, 2019-02-17:
Yeah that seems logical. I just looked through the PT100 documentation and it indeed is intended for 5v. 
https://e3d-online.dozuki.com/Wiki/E3D_PT100_Amplifier_Documentation
Based on the documentation for the INA826AIDGKR which is the amp chip it can accept a minimum voltage of 2.7. I guessing that we could build an alternative temperature table for 3.3v operation.

landodragon141, 2019-02-17:
So after looking at the temp table it looks like every ~0.04v equals a 10C delta. If you could do some measurements we can determine how we need to transform the existing table.

khomouda, 2019-02-17:
I DON'T THINK MY DMM PROBE IS ACCURATE ENOUGH FOR MEASURING an aluminum block, how would you think measuring is accurate

landodragon141, 2019-02-17:
Well it just needs to be close enough to establish how it differs from readings at 5v. Heck I'd measure the temp with a meat thermometer taped to the block with aluminum tape. (I may or may not have done that before :D ). Honestly I think you'll have a more difficult time accurately measuring the voltage than the temp. Anyways just sharing some thoughts on how this could be accomplished in the easiest way.

khomouda, 2019-02-17:
no, the voltage is easier from the terminal coming from the thermocouple and its ground, but my DMM probe is not accurate when measuring at high temperatures from the block

landodragon141, 2019-02-17:
You don't need high temps. Measure at 40, 50, 60 etc

khomouda, 2019-02-17:
up to which temp?

landodragon141, 2019-02-17:
The table at 5v is completely linear so high temps aren't necessary.

landodragon141, 2019-02-17:
As high as you want. As I said we are just trying to establish a pattern.

khomouda, 2019-02-17:
ok

khomouda, 2019-02-17:
here you are sir
[PT100 @3.3v.txt](https://github.com/MarlinFirmware/Marlin/files/2872852/PT100.%403.3v.txt)

AnHardt, 2019-02-18:
Tried to recalculate table_20 from the theoretical values, 
[PT100.zip](https://github.com/MarlinFirmware/Marlin/files/2873547/PT100.zip) For this table the PT100s resistance is assumed to be linear, instead of taking the exact values from literature. That makes the small difference to the original table_20.
If amplifier board and ADC have the same voltage, regardless of if 3.3 or 5 or any other value, the table is the same. Voltage shorts out of the equation.

khomouda, 2019-02-18:
Thanks for the reply, but I don't understand it, the reported temperatures by the printer were at least 50 to 100 degrees more than the measured by my DMM, so the table 20 is completely wrong with 3.3v

AnHardt, 2019-02-18:
> the reported temperatures by the printer were at least 50 to 100 degrees more than the measured by my DMM, so the table 20 is completely wrong with 3.3v

No. It means the theory what table_20 is made with does not match **your** system, or, what is the same, **your** system does not match the theory.

An incomplete list of possible errors - ordered from PT100 to ADC:
* PT100 is not a PT100. Verify with Ohm-meter
* Long, thin, broken, shorted wires to the sensor.
* R1 and/or R2 are not 2K2.
* Voltage over R1, PT100, R2 is not the same as ADC reference voltage.
* R5 is not 5K49. What gives an other gain of the amplifier.
* Amplifier IC is an other type.
* Long, thin, broken, shorted wires to the ADC.
* Amplifier-IC gets an other Voltage than the resistors  and the ADC..
* Additional pull-up -down resistor between IC-output and ADC. (Either external (normal Temperature pin on the board) or internal (is it possible to activate the internal resistors and analog input at the same time?))
* Different ADC reference voltage. 
* Is GND the same everywhere.
* Inperfect contact of plugs / cold soldered contacts.

* right temperature sensor for the DMM
* batteries of the DMM
* uncalibrated DMM
* bad thermal contact of the sensor to the heaterblock
* too fast - different temperatures **in** and **on** the surface of the heaterblock.
* external sensor not isolated against environment.

* wrong interpretation of the table (NTC vs. PTC for example)

khomouda, 2019-02-18:
That same PT100 was working perfectly on the same printer on RAMPS14 (was on A11) before I upgraded to Re-ARM (now on A9 but mentioned in pins file to be 3) and taking voltage and ground from the adjacent pins in AUX2 (3.3v)

AnHardt, 2019-02-18:
[image]
The RE-ARM makes relatively high effort to keep the analog voltages separate from the digital ones.
It could make a difference **where** you take the 3.3V and/or GND for the amplifier board. Analog GND is exposed on the RE-ARMS temperature pins. Don't know where to find analog 3.3V exposed.

Bob-the-Kuhn, 2019-02-18:
I'm a bit confused.  What pin is the PT100 amplifier tied to?

The PT100 amplifier doesn't have enough muscle to drive a 4.7k pullup resistor when powered from 5V.  Dropping it to 3.3V makes it worse.  It must be tied to an analog in that does not have a pullup on it.

To make matters worse the analog inputs are VERY noisy.  The filter helps but it still tends to bounce. T0, T1 & T2 on the Re-Arm card are the best of the analog inputs.

I gave up trying to use the PT100 amplifier on my DUE based system.  I'm currently running the PT100 thermistor/sensor directly into an analog input and have changed the pullup from 4.7K to 1K.  The down side is that only 2 Ohms of wiring resistance will show up as a 5 degree C fixed positive offset.  That really isn't a big deal since you normally need to do some test prints to find the temperature sweet spot every time you switch to a new spool.

AnHardt, 2019-02-18:
There is no way to connect the amplifier to T0-T2. The 4K7 pullups do their job - the temps will be off. (except you remove the pullups.)
But taking AGND from there could be a good idea.
----
Had relative good experience with PT1000 and 4K7 pullups.
Best experience with PT100 i got with a SPI interfaced MAX31865-board with its own ADC (still not in marlin).

khomouda, 2019-02-18:
Thanks for your input, I connected it to A9 in AUX2 together with a 100uF cap according to issue #7646 and called it 3 in pins file (never knew why).
so what is your suggestion? the NTC thermistors did not work at all with PETG, may be are too high of temp difference to the low side, so that it does not melt the filament at 270 deg

khomouda, 2019-02-18:
I will try using AGnd when I go home, I also ordered MAX31865-board just in case it is supported in Marlin

khomouda, 2019-02-18:
I tried using AGROUND in J8 but same response rocket increase in temp reading till it halts the printer, although it never can be that high in this few seconds

Bob-the-Kuhn, 2019-02-18:
My suggestion is:
1. Dump the PT100 amplifier entirely
2. Connect the PT100 thermistor to T0, T1 or T2 on the Re-Arm board
3. Set J7, J8 or J10 as appropriate
4. Highly recommended: replace the 4.7K pullup with a 1K pullup.  This gives better resolution.

I have the advantage of several bad assemblies where I can salvage SMD parts.

___

The big advantage of the PT100 amplifier is resolution.  At 210C the PT100 amplifier has a resolution of about 1.25 degrees.  The 4.7K pullup would be 13 degrees.  The 1K pullup would be 3.5 degrees.

---

> called it 3 in pins file (never knew why).

Per this diagram,  Analog Input 3 is on AUX 2 (pin 63 in the old system, P0_26 in the current system).  It won't have a pullup resistor on it so it is a reasonable choice for the PT100 amplifier input **IF** you can get the PT100 amplifier to behave properly on 3.3V.

Pins that can be used as either analog inputs or digital I/O have two different names.  In this case it is "3" if it is used as an analog input.  If usesd as digital I/O then it is called "P0_26 ".

[image]

---

Another alternative (that I have not tried) - use a 3.0V 5% Zener across the PT100 amplifier to guarantee that it'll never exceed the 3.3V input spec on the analog input.  The downside is that a new thermistor table would be needed.

khomouda, 2019-02-19:
[image]
I did what you suggested (as in picture) and connected the PT100 directly to T0, and changed #define TEMP_0_PIN          0   // A0 (T0) - (67) - TEMP_0_PIN
 but it is giving 1 degree at room temperature.
note that the PT100 is a 3 wire and I am connecting one of the wires and ground

landodragon141, 2019-02-19:
For a 3 wire pt100 you can treat it like a 2 wire. 2 of the wires should have almost no resistance between them. These two should be connected together.

closed 2019-02-20 by khomouda