MarlinFirmware/Marlin #13192: Good afternoon E0 and E1 are not working

xSCIFERx, 2019-02-17:
Good afternoon yall,

Got an issue with my A8-M, I got the BLTouch working, got hotends working as far as temps go and she homes X,Y,Z just fine.

The issues are when I go to print the extruder motor(s) does not activate, not even when set to PLA hotend 200 and bed to 50, I tried activating the E0 or E2 nothing happens the screen shows that its moving and pronterface but motor is not moving.

I've tried forcing the PLA and she comes out fine so it's not clogged, Do I post my config and config adv files?

I've also seen that the screen shows E0, E1, and E2

Anet A8-M
KFB2.0 board
Marlin 1.9
25A PSU
Direct Drive E0 and E1

`// This defines the number of extruders
// :[1, 2, 3, 4, 5]
#define EXTRUDERS 2

#define TEMP_SENSOR_0 5
#define TEMP_SENSOR_1 5
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_4 0
#define TEMP_SENSOR_BED 5

#define PREVENT_COLD_EXTRUSION // not commented

//#define DISTINCT_E_FACTORS

/**
* Default Axis Steps Per Unit (steps/mm)
* Override with M92
* X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
*/
#define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 80, 4000, 300 } // Was 100, 100, 400, 100 but using 20 teeth gears for X&Y
//#define DEFAULT_AXIS_STEPS_PER_UNIT {80, 80, 400, 100, 100 } // I added it
/**
* Default Max Feed Rate (mm/s)
* Override with M203
* X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
*/
#define DEFAULT_MAX_FEEDRATE { 400, 400, 8, 50 }

/**
* Default Max Acceleration (change/s) change = mm/s
* (Maximum start speed for accelerated moves)
* Override with M201
* X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
*/
#define DEFAULT_MAX_ACCELERATION { 2000, 2000, 100, 10000 }

// @section extruder

#define DISABLE_E false // For all extruders
#define DISABLE_INACTIVE_EXTRUDER true // Keep only the active extruder enabled.`

BTW not sure if it matters but I also have octoprint setup with a RPI.

What I've done:
swapped cables for X and Y to E0 and E1, in the panel I was able to get both E0 and E1 to move, plugged that back to normal, I'm not sure on what to do next, the only thing I can think of is that the driver is DOA and/or coding but just guessing now smh. 

Please let me know what other info yall need.
Thank you an advanced.

Best regards.

revilor, 2019-02-18:
Did you adjust the motor current for all the stepper drivers? Did you try to use the X and Y stepper drivers for E0 and E1 to check if it's an issue with the drivers?

xSCIFERx, 2019-02-18:
I did not adjust motor current for any of the motors, I'm using one of the Marlin1.9(Not Marlin1.9BugFix) then I made my changes to get BLTouch to work, got the dimms about 95% right. 

I did not think of changing them as X, Y, and Z are working. (Not sure what to change) 
As far as swapping drivers, I've tried that before I flashed to Marlin with all heats turned up PLA 200 HH1&HH2 and 50 HB so with that said, I then Flashed to Marlin for the safety of them thermos and what not.

I've even tried #define PREVENT_COLD_EXTRUSION and commented.

Working on a fresh copy of Marlin1.9BugFix and adding them as I go down the list of changes I've done with Notepad++ but this starting over S*&^ sucks when everything else seems to be working right but the extruders smh

xSCIFERx, 2019-02-18:
The issue is not the driver hardware just swapped them with a working driver from X and Y.

So it's not the cable, driver so the only other thing I can think of is the OS settings.

Last night I took the time to get a copy of Marlin2 so I'm gonna test with that and keep y'all posted.

Could you be so kind to point me to the right direction as in how to properly set the currents for the motors.

xSCIFERx, 2019-02-18:
Update:
After flashing Marlin2.0 with my changes of dimm's and what not, she now, after getting the right temps, the E0 now works, inverted but works (Easy fix) I did not check to see if E1 is working but after seeing that E0 is now working I can fine-tune the rest, and the screen is on but blank, not showing anything but still lets me connect via USB.

I'd like to close this Ticket as the fix was upgrading to Marlin2.0 

I would really appreciate being pointed in the right direction to learn about the currents going to and from the motors, thank you in advance.

revilor, 2019-02-18:
That depends on the stepper drivers you are using. The most common are

 * [A4988](https://www.pololu.com/product/1182)
 * [DRV8824](https://www.pololu.com/product/2131)
 * [DRV8825](https://www.pololu.com/product/2133)

All those pages above have a section on how to set the motor current. You can also check [this video](https://www.youtube.com/watch?time_continue=1&v=89BHS9hfSUk). 

As your extruders are working now you probably can leave everything as it is. I asked about the motor current because it could have been a possible cause of your problem if the motor current would have been too low. But they are working now so I would keep it that way :)

xSCIFERx, 2019-02-19:
I do appreciate all this information you provided.

I'd like to be able to do a bass test so that way if I have future issues I'll be able to diag this issue quickly.

Thank you.

closed 2019-02-19 by xSCIFERx