MarlinFirmware/Marlin #13194: Question for Bob the Kuhn

n9jcv, 2019-02-18:
Bob, I am sorry I did not see any other way to contact you.  I am new on Github.
You previously posted code, that I am really interested in;
https://github.com/MarlinFirmware/Marlin/issues/8519#issuecomment-346986901

The link to the code is now dead.  Is this something you still have that you can make available for me?
Sorry if I should not have posted here.  Please let me know where this type of post should go.

Thanks
Bruce

Bob-the-Kuhn, 2019-02-18:
What are you wanting to do?

Since that post the endstop logic has been re-written at least once.  Best to start with a description of the desired functionality.

n9jcv, 2019-02-18:
I am working with 1.1.9. I need to have a min endstop for E.  E is a 4th axis and not an extruder for me.

Thanks
B

Bob-the-Kuhn, 2019-02-19:
Try these files.

[E_MIN_endstop.zip](https://github.com/MarlinFirmware/Marlin/files/2878482/E_MIN_endstop.zip)

boelle, 2019-02-20:
Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

n9jcv, 2019-02-20:
thanks for the files Bob, I will try them out this weekend :)

n9jcv, 2019-02-22:
Bob, 
I have tried your files, and looked at some others and made all the changes I could see necessary.  I am using a MKS GEN L 1.0 board.  I am trying to use pin D5.  M43 shows E Min for D5.  M119 shows Emin.  I can change Emin M119 if I change the E_MIN_ENDSTOP_INVERTING from true to false and recompile it changes from open to triggered.  However when I use a wire from D5 to Ground, either open or connecting, it does not change the state from open/triggered.  

I feel like something is still missing.  Here are all my changed files for this change.  If possible could you see what I might have done wrong or what I still need to change please.  At this point I do not have a printer, just testing this marlin code 1.1.9 on the board.  
[11 Marlin Emin.zip](https://github.com/MarlinFirmware/Marlin/files/2892190/11.Marlin.Emin.zip)

Thank you very much for your assistance
Bruce

n9jcv, 2019-02-22:
Just a little update   with M42 P6 S0 and M42 P6 S255 I can change the pin and the M119 changes from open to triggered   So the only issue I have is when I have a physical wire/switch, the state does not change.  I think I missed some code in Marlin, but I can't see where.
Thanks

n9jcv, 2019-02-22:
Update, I just found it.  I had a typo from a copy/paste I did;
UPDATE_ENDSTOP_BIT(X, MIN);
should have been 
UPDATE_ENDSTOP_BIT(E, MIN);

Now it triggers with a switch     Thanks Bob!

closed 2019-02-22 by n9jcv